Ultimaker/Cura #20242: slicing failed, large object

JonnyPhenomenon, 2025-02-05:
### Cura Version

5.9.0

### Operating System

windows 11

### Printer

anet a8

### Reproduction steps

My design file was created in sketchup 17, 100 times scaled up so it could hold the mesh. its a pretty low poly mesh, but whatever.  I scaled it back down within cura, and moved it back to the center of the print bed. clicked slice, and got the slicing failed message. derp derp.

### Actual results

it crashed

### Expected results

not crash

### Add your .zip and screenshots here ⬇️

[led wheel.zip](https://github.com/user-attachments/files/18677056/led.wheel.zip)

GregValiant, 2025-02-05:
Thanks for the report.
I can duplicate the problem with the "Normal" profile.  The model will slice with the "Draft" profile but when I set it to 0.15 layer height it bugs out.

This is not the first report of this so I'll mark this as a duplicate.  It's a simple model and may be a good example for the Cura Team to use for debugging (@HellAholic ?).

HellAholic, 2025-02-06:
[AA8_model_repair_led wheel.zip](https://github.com/user-attachments/files/18683818/AA8_model_repair_led.wheel.zip)

It's a model repair one. Just did a quick model repair and it slices. Model after repair still has some issues but can't do much more with a quick repair tool. 
[image]

GregValiant, 2025-02-06:
I didn't get a notice from MS 3D Builder (which is usually pretty good).
This is from the other place I use.
-> Analyzing file
--> 0 Naked edges  (?)
--> 0 Planar holes  (?)
--> 0 Non-Planar holes  (?)
--> 0 Non-Manifold edges  (?)
--> 0 Inverted normals  (?)
--> 0 Duplicate faces  (?)
--> 0 Degenerate faces  (?)
--> 0 Disjoint shells  (?)
-> Repairing: 100.00%
-> Converting back
-> File repaired
Time needed for repair (ms): 1195
--> Vertex count changed from 2444 to 2994 (+550)
--> Triangle count changed from 4920 to 6020 (+1100)

No errors and then it increases the vertexes by almost 20%.