Ultimaker/Cura #20244: doesn't work m83

Holyfriend, 2025-02-06:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Custom

### Reproduction steps

If Settings => printer => printer setting => G-Code flavor = "Merlin" and checkbox "absolute/relative extrusion" on.

### Actual results

Cura generate Gcode only for M82.
Checkbox "absolute/relative extrusion" doesn't work.

If switch Settings => printer => printer setting => G-Code flavor = "RepRap".
Checkbox "absolute/relative extrusion" is work, and cura generate gcode m82 or m83 !!!

### Expected results

Need work checkbox "absolute/relative extrusion" for G-Code flavor = "Merlin".

### Add your .zip and screenshots here ⬇️

[Gcode.zip](https://github.com/user-attachments/files/18683364/Gcode.zip)

Holyfriend, 2025-02-07:
My printer runs on Klipper, but it doesn't matter.
What is important is that when G-Code flavour = ‘Merlin’ (default) - checkbox ‘absolute/relative extrusion’ stops working.
Regardless of the checkbox position - gcode will be generated via m82 (which assumes accumulation of errors during printing).

Check it yourself, switch the checkbox on and generate any gcode, then switch it off and compare the files - there will be no differences in the gcode structure, although there should be, since a different extruder mode is selected.

And the strange thing is that if you select G-Code flavour ‘RepRap’ - cura starts generating gcode correctly in m83 mode (it is more accurate).

Holyfriend, 2025-02-07:
Reinstalling cura did not help.
In cura.log when changing the checkbox the change ‘-s machine_use_extruder_offset_to_offset_coords=’False/True‘’ occurs, but it does not affect Gcode itself (unless G-Code flavour = ‘RepRap’ is selected).

GregValiant, 2025-02-08:
I still can't duplicate this.
This is for my Ender 3 Pro running Marlin firmware and Cura is set to Marlin.  You can see that the M83 is there and the code is corrent.
```
;---------------------[End of StartUp]
M83 ;relative extrusion mode
M106 S0
;LAYER_COUNT:250
;LAYER:0
M117 1/251 | ET 2h30m
M118 1/251 | ET 2h30m
M106 S0
M204 S1000
M205 X8 Y8
G0 F7200 X167.117 Y145.144 Z0.2
;TYPE:SKIRT
G1 F3000 X165.806 Y145.257 E0.06796
G1 X165.17 Y145.27 E0.03285
G1 X99.388 Y145.269 E3.39736
```
I asked previously - what printer definition did you use to install your printer in Cura?  Was it as a "Custom FFF" printer or did you have another file?

Holyfriend, 2025-02-08:
Thanks, I figured it out, the problem was on my side.
Everything is working correctly now.

GregValiant, 2025-02-08:
Thanks for getting back.  There was about a 99.9% chance that it would be a local problem.
I'll go ahead and close this.

closed 2025-02-08 by GregValiant