Ultimaker/Cura #20251: Flsun v400 connection not possible

Moi76290, 2025-02-07:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

FLSUN V400 (with clipper)

### Reproduction steps

I have added my FLSUN v400 as a printer on Cura 5.9.0
I checked that my printer and my computer were on the same Wifi network
I went to ‘Manage printers’

### Actual results

The ‘Connect Moonraker’ option doesn't appear and I can't connect my printer.

### Expected results

The ‘Connect Moonraker’ should appear
With Cura 4.13.1 I don't have this problem

### Add your .zip and screenshots here ⬇️

With Cura 4.13.1

[image]
With Cura 5.9.0

[image]

Asterchades, 2025-02-08:
You will need to install the _Moonraker Connection_ plugin from the Marketplace, as you have done with your 4.13.1 installation. Cura has no native support for any network printers other than Ultimaker - all such support is provided via plugins.

Moi76290, 2025-02-08:
Thanks a lot, now it's OK

closed 2025-02-08 by Moi76290