Ultimaker/Cura #20255: jittering in slicer cause a message that says unable to slice

Tri-pointMechatronics, 2025-02-09:
### Cura Version

5.8.0

### Operating System

windows 11

### Printer

Elegoo Neptune 3 max

### Reproduction steps

1. set up the file in the configuration needed (in the middle)
2. added the "support blocker" 
3. hit slice

### Actual results

error message "unable to slice"

### Expected results

It should have sliced

### Add your .zip and screenshots here ⬇️

https://we.tl/t-hdOg1YaAGr

[image]
the reason I think this is happening is because the XYZ potion is flipping between 0.4667 and 0.4668, but I don't know for sure

GregValiant, 2025-02-11:
Thanks for the report.
After loading the project the "Slice" button never appeared.  I had to "Import Models" and then everything seemed to work.
A few of the models need repair.  The errors could be severe enough to cause the slice to fail.

[image]

Here I've thrown out all the models but the one, and still no slice button.  I have no idea what's up with that.

[image]

I suggest you remove the models you aren't currently slicing in order to recover some resources and then try again.  You should also check the models and find the ones that need repair and take care of that.

The Cura team will take a look.

GregValiant, 2026-09-08:
I'll close this.

closed 2026-09-08 by GregValiant