Ultimaker/Cura #20258: impossible de mettre le hauteur d'impression a 40CM

lebuffalo, 2025-02-10:
### Cura Version

5.9.0

### Operating System

windows 11

### Printer

artillery sidewinder X2

### Reproduction steps

dans les paramettres de l'imprimante la valeur Z (hauteur) dois etre de 400mm mais sur le plateau cela ne change pas et les pieces ne peuvent pas etre plus haute que 200mm

### Actual results

hauteur (z) sur le plateau reste a 200mm

### Expected results

la hauteur (z) devrais passer a 400mm

### Add your .zip and screenshots here ⬇️

[image]

[image]

GregValiant, 2025-02-11:
Thanks for the report.
This came up recently with another printer.  I'm unable to locate that report, but I think it was also an Artillery X2.
Cura does not have a definition file for that printer.  I loaded an X1 (300 x 300 x 400 build volume) and everything worked as it should.
What did you use for a printer definition when you installed the printer in Cura?

[image]

GregValiant, 2025-02-14:
Check the "Vertical Scaling Shrinkage Compensation".  It can affect the build height.  The default is 100.0.

closed 2025-02-25 by github-actions[bot]