Ultimaker/Cura #20267: Failed to slice features

dpaul6463, 2025-02-12:
### Cura Version

5.9.0

### Operating System

Windows 10 pro

### Printer

Coocheer

### Reproduction steps

My round part has 4 small holes (radius 1.5mm) through the bottom (2mm thick) at an angle of 60 deg to the perpendicular. 
The STL is made with Sketchup.

0.4mm nozzle
0.15mm layer height

### Actual results

With the part oriented with the bottom down, "Prepare" shows the features but the "Preview" does not show the holes and it also does not print them. It's the same problem with and without supports.

HOWEVER. If the part is oriented on edge it DOES build the features (but that doesn't work for other reasons).

### Expected results

I expect the features (0.15mm radius holes at 60 deg) to be sliced and printed per the STL design.

### Add your .zip and screenshots here ⬇️

[Pinhole base.zip](https://github.com/user-attachments/files/18760487/Pinhole.base.zip)

Asterchades, 2025-02-12:
>  The STL is made with Sketchup.

This immediately points to the reason. SketchUp likes to make holes with inverted normals. I don't know why, but it does. The default colour scheme of white and not-quite-white doesn't make this particularly obvious when the areas get a little bit shaded, though, so often times you won't learn that it's done this until it's time to make a post somewhere asking what's gone wrong.

Here's your file loaded into SketchUp 2017 with the back faces set to be burgundy instead, so that they stand out clearly:

[image]

Note that the main hole in the centre is white as it should be, but the holes that disappear are burgundy. They're reversed, so they're defining solid shapes - just without the end caps. Slicing this, layer by layer, they just look like solids and are subsequently getting filled in, but as you rotate the model they suddenly have open sides and correctly print open.

Good news is that this is otherwise a good model, so it's a quick and easy fix in SketchUp: right click on any of the white surfaces and select _Orient Faces_, and everything will be corrected. Note that if it wasn't an otherwise good model this could cause it to turn into an absolute mess as SketchUp attempts to determine what is and isn't "facing" in the same direction.

Should you not wish to fix the model, you could use the _Mesh Tools_ plugin from the Marketplace. By right-clicking on the model and using the _Fix model normals_ option it is doing similar to the above, which again works just fine as it's an otherwise good model.

You could also disable the _Union Overlapping Volumes_ mesh fix in this particular case, but that should not be treated as a panacea for all models showing this issue as there are other common faults it will correct.

GregValiant, 2026-09-08:
I'll go ahead and close this.

closed 2026-09-08 by GregValiant