Ultimaker/Cura #20273: AS SOON AS I LOAD A stl, PROGRAM FREEZED AND HAS TO BE CLOSED

LORDDOUGLASO, 2025-02-13:
### Cura Version

5.9

### Operating System

wINDOWS 11

### Printer

iMEGA x aNYCUBIC

### Reproduction steps

LOADED stl FILE * MANY DIFFERENT ONES TO TEST ) 
pROGRAM FREEZES AND WONT RECOVER
 

### Actual results

pROGRAM FRREEZES

### Expected results

Restart program 

### Add your .zip and screenshots here ⬇️

program version sucks

GregValiant, 2025-02-13:
Thanks for the report.
If you have USB devices installed (like headphones, earbuds, etc.) and if one or more is internally configured as a Serial Device, then it's possible that Cura is constantly polling the device thinking that it's a printer.

Go to the MarketPlace and click on the "Gear" icon.  Scroll down to "USB Printing" and disable it.  Restart Cura.
Let us know if that works for you.

GregValiant, 2026-09-08:
I'll close this.

closed 2026-09-08 by GregValiant