Ultimaker/Cura #20276: Oops! We encountered an unexpected error during your slicing procesS..CANNOT SLICE

ivatman, 2025-02-13:
### Cura Version

5.8.1

### Operating System

windows 11

### Printer

ender 3

### Name abnormal settings

yes, it crahses with any profile

### Describe model location

no, it doesnt slice

### Describe your model

it is a half part of a rock

### Add your .zip here ⬇️

MODEL DOENST SLICE, MESH IS OKAY

GregValiant, 2025-02-17:
I don't have access to download from there.

ivatman, 2025-02-17:
How do I share the file with you?

ivatman, 2025-02-19:
[export-20250217-121040.zip](https://github.com/user-attachments/files/18856875/export-20250217-121040.zip)

ivatman, 2025-02-19:
> I don't have access to download from there.

[export-20250217-121040.zip](https://github.com/user-attachments/files/18856891/export-20250217-121040.zip)

GregValiant, 2025-02-19:
I downloaded the file and sliced it with 5.8.1 and 5.9.0.  I'm unable to duplicate the problem.
Maybe digging deeper will give a clue.  Use the "Help | Show Configuration Folder" command and zip the cura.log file and post the zip folder here.
This is in 5.8.1.
[image]

closed 2025-03-05 by github-actions[bot]