Ultimaker/Cura #20277: Print just stopped after about 12 hours

nigelthecoder, 2025-02-14:
### Cura Version

5.9.0

### Operating System

MacOS Sequoia

### Printer

Creality Ender-3 V2 Neo

### Reproduction steps

I started a print that was going to take about 13.5 hours.
After about 12 hours it just stopped. No status in Cura at all. The Pause and Abort buttons greyed out. No print progress showing anymore.
Printing from my Mac mini over USB.
You wont let me add an STL file huh?
Or gcode - well that about does it for what I could have sent you.

### Actual results

not finished

### Expected results

a nice box

### Add your .zip and screenshots here ⬇️

no further comments other than this is a protty silly reporting thing that can't accept the STL or GCODE files

GregValiant, 2025-02-14:
Thanks for the report.
Github rules are that all files posted here must be zipped.  A couple of exclusions are ".txt.", "JPEGS" and a couple of others.
I don't think we need a file for this.

USB Printing is glitchy at best.  It is kept in Cura because some older UM printers require it.  There was a reason that UM moved away from USB printing.  The reason is still valid today.  Every printer, operating system, computer mainboard, firmware, USB port hardware, port drivers, etc. etc. vary.  Trying to take into account all the combinations that can occur is a useless endeavor.

My own experience is that when attempting over-night prints, too often, Microsoft would push out an update that would cause my computer to restart.  That kills the print, and when the printer comes back up it re-establishes the 5volts to the printer mainboard which causes the printer to reboot.  At that point recovering the print is a very difficult chore.  It can be done, but it never really works out right.  There can be serious weakness at the restart layer, there can be layer shifts in the X or Y of up to 0.5mm.
I suggest you abandon trying to use the USB to print.  If you want to print remotely then look into something like OctoPrint which has a lot of support and a lot of users.

I'll mark this as a duplicate.  I'm thinking about closing this as it has been re-hashed many times.  Let me know what you think.

nigelthecoder, 2025-02-14:
Thanks for the reply.
I spent a lifetime as a programmer and I'm very familiar with the Evil
Empires push policy (worked there 89 - 96).
These days I use a Mac mini, but same issue I think sometimes.
Yesterday I had a USB print stop after 30 mins - maybe because I was also
doing paper prints, although these were via AirPrint not USB.

Anyway - I'm going back to uSD card temporarily , and then back to
OctoPrint on Raspberry Pi Zero 2W which I set up recently and seems to work
quite wel.

Regards,
Nigel

GregValiant, 2025-02-15:
Octoprint is a much better choice than Cura's USB printing.

Back-in-the-day I wrote apps for communicating with different sorts of 'serial' lab equipment and test machines.  They all had their peculiarities and it was an odd thing when one script would work on more than one machine (even if the manufacturer was the same for both machines).

There are a lot of Cura contributors who are experienced programmers (and of course there is the Cura team).  Whoever was able to fix USB Printing would be a hero.  As you noticed, no one has bothered to pick up the ball on that.

I'll go ahead and close this.  This bug has been reported many times and there are workarounds.  I hope you understand.

closed 2025-02-15 by GregValiant

nigelthecoder, 2025-02-16:
That you so much for taking the time to reply to my emails.
It's a rare thing these days to have actual contact with a human entity.
I'm printing a box for my RasPi Octoprint. In the man time I'm doing the
uSD card shuffle,

Live long and prosper,

Nigel