Ultimaker/Cura #20279: Print volume to small

Eljesah, 2025-02-14:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

Ultimaker shows a much smaller print volume then the 330 mm X 240 mm X 300 mm. The height is a lot smaller in Cura.
I have checked the settings and everything is correct there.

I can´t print because Cura says the part is too big for the printer even if it fits easily, because of the wrong small size in Cura.

[image]

### Actual results

No printing possible

### Expected results

Show real print volume size

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-02-14:
Thanks for the report.  This has come up a couple of times lately ( #20258 is one).
Please post a zipped project file ("File | Save Project") and someone will take a look.

Eljesah, 2025-02-14:
I just downloaded Cura 5.6.0 and linked my printer and the print volume appeared in the right dimensions. I then clicked sync printer to have all of my materials.. in Cura and the print volume decreased to the small wrong size after sync was finished. I can´t get it back to the original size.

Eljesah, 2025-02-14:
Maybe important to mention. The build volume on the screen jumps back to normal fullsize, when I click on the printer in Cura. But the full build volume is only displayed for less then a second and it goes back to the smaller size

GregValiant, 2025-02-14:
Maybe @HellAholic has heard of this happening.  

We still need a project file.  Even better, use the "Help | Export file for Technical Support" command then zip and post that file.  It will have your log file in it.

HellAholic, 2025-02-14:
Under the Print settings, check the value of 'Vertical Scaling Factor Shrinkage Compensation'
this value for PLA should be '100.1'
if the value is for example 600% then you can get that reduced height for the build volume.

[image]

GregValiant, 2025-02-14:
I gotta take that off the list of "things I didn't know".

Eljesah, 2025-02-14:
@GregValiant my setting is correct at 100.1, but I still get the small print volume. Here is the zipped file.
Thank you

[UMS5_Test01.zip](https://github.com/user-attachments/files/18805787/UMS5_Test01.zip)

Eljesah, 2025-02-14:
and here is the file exported with  "Help | Export file for Technical Support" 
Thank you

[export-20250215-005544.zip](https://github.com/user-attachments/files/18806088/export-20250215-005544.zip)

Eljesah, 2025-02-15:
I tried changing the Vertical Scaling Factor Shrinkage Compensation to 600% and the Z size goes to really small for half a second and jumps back to this wrong scale that I always get. I tried to change the 300mm in Z to only 20mm in the Cura printer setting but it still stays at the same size. X and Y are changeable in the Cura printer settings. I can go smaller or even much bigger and the build volume size in the view reacts to that. But the Z is always the same no matter what I type.

GregValiant, 2025-02-15:
This is what I see when I open the project file:

[image]

Have you tried installing a second instance of an S5 in Cura?  If that doesn't fix the problem, you may have to do a complete re-install of Cura.  Something might be corrupt in your configuration folder.  The technical term is "Stuck on Stupid".  It's something that computers and peripherals seem to be especially good at.

GregValiant, 2025-02-15:
I'm not good at the "What happens during a re-install".
Your custom settings are in the Configuration Folder.  You can find it by using the "Help | Show configuration folder" command.
Better that @HellAholic responds.
I have a guess as to what is happening, but it's only a guess and I don't want to steer you wrong and cause the loss of your customizations.

HellAholic, 2025-02-15:
The range of answers I can provide ends with Cura, if you have questions regarding the other elements that are outside of the scope of this repository, I cannot provide any concrete answers. It depends on many factors is the best answer I can give.

Since you're having the issue in all versions, it does not seem to be related to Cura itself. I can say for sure that there is an issue in one of the components in there but which one and why is not possible to diagnose and resolve here.

Best solution I can provide is to Close all Cura versions, create a backup of your configuration folder `/Roaming/cura` & `/Local/cura`; then delete both cura folders in Roaming and Local, then re-launch Cura, sign in to get your printer but **DO NOT** install the materials and plugins with account sync, install them manually 1 by 1 from Marketplace and see at what point things start going bad.
If you notice things are missing and you would want to get them back then restore the folders from the backups you've made.

Eljesah, 2025-02-16:
Thanks for the support guys. It took a while to do it one by one but it has worked. Plugin "Fisnar Robot Plugin" caused the issue.
I was constantly asked to sync my material and software packages with my account everytime I opened Cura. I then did that (it cause the print volume to decrease again) and deleted "Fisnar Robot Plugin" manually and it worked.

Everything is back to normal now. Thank you very much :)

GregValiant, 2025-02-16:
Nice job chasing that one down.
This doesn't sound like a Cura problem so I'll remove the bug label and close this.  I hope you understand.

Eljesah, 2025-02-16:
Thats fine as long as others with the same problem can read this

GregValiant, 2025-02-16:
It will still come up in searches.
I'm not familiar with that plug-in.  I looked at the author's Git and it doesn't appear to have been updated in a while.