Ultimaker/Cura #20280: Top/Buttom initial layer pattern

krsshtx, 2025-02-15:
### Is your feature request related to a problem?

Printing overlaps in circular model with zig-zag produces bad inner surface.
With circular pattern and complex top form gives excessive contours and bad layer adhesion.

### Describe the solution you'd like

Circular initial pattern for top-buttom would be optimal option with subsequent lines or zig-zag.

### Describe alternatives you've considered

None.

### Affected users and/or printers

Any.

krsshtx, 2025-02-15:
> Just to be clear... There is currently a setting for "Bottom Pattern Initial Layer". Is that acceptable as it is currently implemented? If the bottom is OK then this request would be for something similar for the top. Ex: "Top Layer Final Pattern"?

That's a different thing.
(Added screenshot).

[image]

Asterchades, 2025-02-16:
You can already replicate this on the top using the _Top Surface Skin Layers_ feature. While it's normally intended to be used for a finishing layer, there would be nothing at all stopping you from using it to replace all but one layer on the top: set it such that it's one layer less than your overall _Top Thickness_ and then configure their respective patterns accordingly.

This of course doesn't help for the bottom layers. Currently, as Greg has pointed out, only the initial layer can be changed separately - that would still leave any bridges (experimental mode enabled or no), overhangs, or supported areas. Perhaps what you're thinking of would be best represented by a "bottom" counterpart to the above setting?

krsshtx, 2025-02-16:
> You can already replicate this on the top using the _Top Surface Skin Layers_ feature.

That's really fun idea! Currently I use 2 top surface layers with reduced thickness to improve surface. I beleive it does so. 
But may be it it will be better.

> Perhaps what you're thinking of would be best represented by a "bottom" counterpart to the above setting?

May be it's impossible. If programm cannot distinguish top part from bottom part by separation of flat layers.
I would use concentric for top/buttom, but it produces not really a concenctric result and repeats it

[image]

 (a pic).