Ultimaker/Cura #20282: Overlapping objects print twice

T9Air, 2025-02-16:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Creality Ender 3 v3 se

### Reproduction steps

Take any two 3d models and have them overlap each other. Making them have different infill settings will make it easier to see the bug.

### Actual results

The slicer makes it print one after the other, so that it will print in the the intersection of the 2 files twice.

### Expected results

It should only print in the area of the intersection once.

### Add your .zip and screenshots here ⬇️

This is just an example
[image]

GregValiant, 2025-02-17:
Thanks for the report.
This might be the only problem with "Union overlapping models" or one of the other mesh fix settings.

Post a project file.  I can't duplicate this "on call" but I know I've seen it happen in my own prints.

GregValiant, 2025-02-17:
Right-click on the file in Explorer and select "Send To...| Compressed (zip) folder" and post the zip folder here.

T9Air, 2025-02-17:
[example.zip](https://github.com/user-attachments/files/18833232/example.zip)

GregValiant, 2025-02-18:
Thanks.
For that combination I would turn off "Union overlapping volumes".  With it turned on, both models have top skins and the top skins of the cylinder are completing themselves.  Unfortunately that means crossing the cube while extruding the skin of the cylinder.

I don't see a bug here.  That is expected behavior given two models that occupy the same space and your settings.  By changing the values of "Union..." and "Remove mesh intersection" and "Alternate mesh removal" you will get different effects.  It might be that properly merging the models in a CAD app (or something like MS 3D Builder) would be sufficient.

There are now more than 600 settings in Cura.  There is always a risk of putting settings together in such a way that they don't play together nicely.

closed 2025-02-18 by T9Air