Ultimaker/Cura #20284: My blender model is not working

rstclair82, 2025-02-17:
### Cura Version

5.4.0

### Operating System

macOS Sonoma

### Printer

Maker Pegasus and Wanhao Duplicator i3

### Reproduction steps

I made a splatoon amiibo model on blender for my son and then I saved it.

### Actual results

Cura started bugging out.

### Expected results

For cura to start bugging out

### Add your .zip and screenshots here ⬇️

[image]

it just shows that. can you please fix it?

GregValiant, 2025-02-17:
Thanks for the report.
One thing that can happen is the model was exported in cm's and Cura always assumes mm's.  If you know 1 dimension of the model then with "uniform scaling" turned on you can scale the correct axis to that size.  For example:  You know the model is 25mm tall so set the scaling tool to "Uniform Scaling" and then make the Z = 25mm.  Then you need to place the model.
In Cura you can try right clicking on the build plate and "select all models" and then on the left toolbar select the "Translate" tool at the top and set the 3 axes each to "0".

Here you can see that the model shadow is on the build plate, but in the "Z" box of the translate tool it is at a Z=500 so I can't see the model.  Setting the Z to 0 will drop the model back to the build plate.

[image]

GregValiant, 2026-09-08:
I'll close this.

closed 2026-09-08 by GregValiant