Ultimaker/Cura #20286: objects do not appear

dopasquier, 2025-02-17:
### Cura Version

UltiMaker-Cura-5.9.0-linux-X64.AppImage

### Operating System

ubuntu 24.04

### Printer

ultimakerS5

### Reproduction steps

1 up grade my os to ubuntu 24.04
2 install UltiMaker-Cura-5.9.0-linux-X64.AppImage
3 launch in a terminal cura
4 the object does not appear  

### Actual results

i can not load the model

### Expected results

what can i do ?

### Add your .zip and screenshots here ⬇️

[image]

dopasquier, 2025-02-18:
the object loaded appear 5-10 minutes later.
i can slice and change the setting and print.

Cura works but load a project is very long.

GregValiant, 2025-02-18:
Thanks for the report.
If you have other USB devices plugged into the computer that are internally configured as "serial devices" then Cura will think they are printers and will keep trying to connect to them even though they might be earbuds or headphones.
Go to the Cura Marketplace (button in the top right corner) and click on the "gear" icon.  Scroll down to "USB Printing" and disable it.  When Cura restarts it should be OK.

Let us know if that works for you.

dopasquier, 2025-02-18:
Bonjour Greg
I  use usb device to transfer the file on the printer ‌. thank you for the explainations and the solution to cure it.
It is working very well now !

Thank you very much

Dominique Pasquier

GregValiant, 2025-02-18:
Good that it's fixed.
I'll mark this as a duplicate and close it.  I think there will be a setting in the next release so a user must specifically state that they want USB Printing enabled.  Right now it is enabled by default.

closed 2025-02-18 by GregValiant