Ultimaker/Cura #20298: When using preview the image is awkward

AllanRomanato, 2025-02-20:
### Cura Version

5.9.1

### Operating System

MacOS Sequoia 15.3

### Printer

Creality Ender 3 v3 KE

### Reproduction steps

After slicing my stl, in preview tab, the image totaly bugged.
When I save it to GCODE everything works fine - it prints ok with no issues.

### Actual results

Here the result
[image]

### Expected results

It should show us the sliced image with no extra info like it is.

### Add your .zip and screenshots here ⬇️

No problrms with printing

GregValiant, 2025-02-24:
Thanks for the report.
This is what I call the "Peter Maxx" bug or the "cockroach" bug (impossible to get rid of) and it shows up every once in a while (for example #16175).  Nobody has been able to duplicate this on purpose, and so the cause remains a mystery.  I think it has something to do with Cura's communication with the video drivers on certain systems, but that's a guess.

Yours is pretty good, but this one is my favorite.
[image]

If you do a search here for "Peter Maxx" you will come up with some other examples.
I'll mark this as a duplicate and go ahead and close it.  I hope you understand.

closed 2025-02-24 by GregValiant

fieldOfView, 2025-02-24:
You can set the option "Use layer view compatibility mode" in the General preferences of Cura. This will mitigate the drawing issue, but it will also disable some functionality.