Ultimaker/Cura #20299: add a setting for support interface/underside of overhanging print temperature

iceman2511, 2025-02-20:
### Is your feature request related to a problem?

So i just recently finished a part with support, and obviously its a complete pain to remove, why wouldnt it be, but i noticed that its evem more of a pain than it should be, like the support interface is melted and fused to the print. support structures have gotten better over the years, but they are still a pain to remove, as well as leave a bad surface on the bottom of the print, and with higher temperature filaments, its even harder, or even impossible to remove the support interface from the print.

### Describe the solution you'd like

So id go and do a bit more in depth with my research about support settings, and found this https://www.instructables.com/Best-Support-Settings-for-3D-Printing/ blog/article, though it may be 7 years old in 2025, there is one key thing that they changed which completely changed how easy it was to remove supports, as well as slightly improve quality of the underside of the support, and that is the temperature that the print is printed at, or more importantly, the temperature the support interface, and the underside of the overhang of which is supported.

and while this single parameter is the difference between having clean support removal, and having fused supports on your parts, i havent seen a feature similar to this on the most popular slicing softwares (cura and prusaslicer), this post is a proof of concept, that i hope someone can expand upon, and hopefully make it a viable feature.

### Describe alternatives you've considered

could print at lower temperature alltogether, such as in the article, but it will affect the whole of the print properties which overall make this a bad decision. print speed, layer adhesion, filament temperature requirements etc etc...

### Affected users and/or printers

people with single extruder printers that cant use a different material as their support, or when printing with high temperature filament.

GregValiant, 2025-02-24:
My printer is an Ender 3 Pro with a single nozzle.  I've thought about this problem and went so far as to write a post-processor for changing materials just for the top layer of the Support Interface.  I works well for large flat interfaces, but because there are always 2 material changes involved (pause to switch to interface material and then pause again to switch back to the model material) it is really annoying for anything other than large flat areas.  (For example, it's not good with large horizontal holes that might require 6 or 8 pauses.)
Another (poor) fix I came up with was to simply pause while I "painted" the interface with a magic marker pen.  PETG and PLA don't stick will to the painted interface, but again, there can be a lot of pauses and it was better for large flat areas.

Changing temperatures has it's own drawbacks.
Where should the nozzle be while it waits for the temperature to fall?
There is going to be some oozing while it waits and if the head parks, that oozed plastic will end up on the model.
If it doesn't wait for the temperature to rise, then you will get poor layer adhesion for the walls, infill, etc. until the nozzle comes back up to temperature.

The gcode generated by UM Cura, (and Prusa/Orca/Bambu) don't have any notation that indicates the "first skin over support".  A post processor won't work and the code would necessarily have to be in the Slicer Engine.

Although I use the "Support Interface Material Change" post processor for prints with large flats (and it does work well), the only really good fix for this for "all models" is to go with a multi-extruder printer that will automatically switch to a support material.

iceman2511, 2025-03-25:
TLDR, changing temperature at initial layer model interface works with flat supported layers, workaround is using change at z post processor, and reduce temp at specified layer, no pausing is required, but reduce print speed otherwise underextrusion may happen.

thought id update this post since ive now done some experimentation with this concept. this concept definately works no doubt, unfortunately i dont have any pictures, but i can say that with PETG at 240C, fans at 85% with 2 5015 fans (100% at model inteface), and support interface of 0.2 distance and 55% density, and with no temperature changes when printing the overhanging model over the support, after trying to remove the support, it is impossible to remove, and can be pretty much said the support interface has "bonded" with the model.

but like @GregValiant has said, you can use change at z post processing to change the nozzle temperature at a specific layer, for me i changed it to 205C as i thought any lower would cause significant issues with PETG, like the viscosity of the material. i didnt pause the print when changing the temperature of the nozzle, as it would both take a long time to do when the nozzle is inactive, and also ooze filament/cause underextrusion in the process, and also because the nozzle cools down very quickly when there is filament extruding from it, just from rough estimations, it takes about 15 secs to reduce about 35 degrees, which id say is quite quick, and im using a copper heat block, so people using an aluminium heat block will experience much faster results as aluminium contains roughly 3 - 4 times less heat/energy than copper for same volume. what i also changed when reducing the nozzle temp is the print speed, as ive tuned my print speed for maximum volumetric flow, having a lower temperature means that it will also reduce the maximum flow of my filament, so i did 50% speed. and just input the layers that you want the affect to take place, and thats it.

with these settings, and with the exact same model and print settings, with some pliers to grip the filament, i could remove the support in 2 or 3 large pieces, and have no leftover support pieces left on the underside of the model, as well as have the underside feel super smooth, and also look much nicer than usual.