Ultimaker/Cura #20307: Make infill removable

SudoPseudo, 2025-02-22:
### Is your feature request related to a problem?

Hello.  I want to be able to make infill removable.

For example, I am printing an object with 0 bottom layers.  It would be nice to be able to change the infill settings (like minimum z, x/y distance from the shell) so I can easily just pull the infill out and have a nice shell that is still well supported during the print.

### Describe the solution you'd like

Difficulty removing infill from hollow prints

### Describe alternatives you've considered

I tried creating a hollow print and then adding supports, but they would only generate on the outside of the print.

### Affected users and/or printers

All users who want to print hollow objects

GregValiant, 2025-02-24:
I'm not sure I understand this.
It is always best to give the slicer the model you want to print, rather than one that's "close, except for....".
In this case it seems that you could make the model hollow and with no "floor".  That would allow regular supports to work on the inside.

Cura has a setting called "Infill Support" which I think will do what you are asking for.  With the "Infill Overlap" set to a negative number the infill will not contact the walls.
A downside is that you can't control the airgap between the top of the infill, and the first skin above the infill.  They will be bonded together making the infill difficult to remove.

The model on the left is a hollow cylinder designed with 1mm walls and a 1mm thick top.
The model on the right is a solid cylinder with Cura set for 1mm walls and 1mm thick top skin, and no bottom skins.  The "Infill Overlap" is -0.80.
Regular "Support" will not work on the solid model, because the model always remains a solid.  Cura can be told not to print the bottom skins, but they are theoretically still there, they just get ignored for printing.
[image]

Tylersuard, 2025-03-03:
@GregValiant Thank you for your comment.  My 3d model is super complex, and hollowing it always introduces problems.  I have not been able to hollow it successfully using Meshmixer or fusion360.  I want to print my model as a single layer (similar to vase mode) to save filament weight.

Yes, you are right, setting the infill support parameter to negative would be helpful but I would also need to have the vertical equivalent of that too.  

When I use supports I can remove them pretty easily by setting the minimum Z and X/Y distance from the print, but I can't do that for infill.

GregValiant, 2025-03-03:
The limit for any single-thickness print is a trig function of the LineWidth and LayerHeight.  For most intents and purposes it is an angle about 27° off the build surface.  Once the angle falls below that (approaches horizontal) the print fails.  There is no way around that.  The single line extrusions become too far apart to bond to each other.  That's why there is never a roof on a spiralized print.

What I see here is a niche request.  My guess is that the UM team won't get to this as it will receive a low priority (and may not be possible at all).  That leaves "workarounds" for the foreseeable future.

Have you tried using "Lightning Infill"?  It might be the best bet, but it would not be removeable.  Left in place it would not weigh much.