Ultimaker/Cura #20308: Cura Print via cloud doesn't send print job to Digital Factory

adrianozandonella, 2025-02-24:
### Cura Version

5.9.0

### Operating System

Windows 10 Pro

### Printer

Ultimaker S5

### Reproduction steps

Cura Print via cloud doesn't send print job to Digital Factory

### Actual results

Cura Print via cloud doesn't send print job to Digital Factory

### Expected results

Cura Print via cloud doesn't send print job to Digital Factory

### Add your .zip and screenshots here ⬇️

Cura Print via cloud doesn't send print job to Digital Factory

ThorsenRune, 2025-02-24:
Same problem after updating firmware.

adrianozandonella, 2025-02-25:
Yes, did firmware update yesterday. S5 is on the latest version 9.0.2
Network connection of S5 shows connected.
Printer is visible in Digital Factory.
When clicking "Print via cloud" in CURA loading bar appears, but no uploading progress happens,

ThorsenRune, 2025-02-26:
[image]
Digital factory is stuck in uploading for weeks