Ultimaker/Cura #20310: Build volume hight keeps resetting

Lazy13b, 2025-02-25:
### Cura Version

5.3.1

### Operating System

win 10 home

### Printer

Ender 3 v2

### Reproduction steps

inserted a model

### Actual results

the build plate height shrinks to about 10mm

### Expected results

nothing

### Add your .zip and screenshots here ⬇️

nothing to add

GregValiant, 2025-03-02:
Thanks for the report.
I keep looking at this report and I'm not really sure what to say.  There are two settings that can affect the build height.

I'm not sure if "Material - Vertical Scaling Factor Shrinkage Compensation" is in 5.3.0, but it will affect the build height when the setting is a really high number (like 1000).

If you have the Print Sequence set to "One at a Time" then the build height will be shown as the "Gantry Height.

You should update to a more current version of Cura.  5.3 is at least 8 versions behind.  For that reason, I'll go ahead and close this.  I hope you understand.

closed 2025-03-02 by GregValiant

Lazy13b, 2025-03-02:
Thanks for your help. It was the "one at a time" setting. I'm not how or when it got selected, but that was the issue. I feel like a dummy. 😂 But thanks again for your reply! 

Yahoo Mail - Email Simplified