Ultimaker/Cura #20320: Cant make a pick/plectrum standing and flat. Not sure why just yet.

Juri3D, 2025-02-26:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Ultimaker 2+ Connect (Europe)

### Name abnormal settings

Support

### Describe model location

Center

### Describe your model

Guitar pick/plectrum, the small plastic thing you hit the strings on the guitar with.

### Add your .zip here ⬇️

[UM2C_standard_351_pick+standing.zip](https://github.com/user-attachments/files/18987632/UM2C_standard_351_pick%2Bstanding.zip)
Here is the original pick/plectrum that I downloaded, it worked, but then when I modified it there was some sort of problem. Sending both files. The smooth file is the problem file.

Juri3D, 2025-02-26:
Flat worked. Its the standing that is causing problems.

GregValiant, 2025-02-26:
Thanks for the report.
Any model that comes to "knife edges" is going to have trouble slicing.  You can try going to a smaller nozzle, but the edges of the model get down to 0.0 thickness at the end and so they just can't slice.
I would also think that the performance of the vertical print would depend a lot on layer adhesion.  I really don't see it holding up during use.  The horizontal print at least has a chance.
The model on the right is 0.75mm thick.  You can see on the left that the width is 0.0 at the top.  You can try lowering the "Minimum Line Width", but it won't be enough.
[image]

GregValiant, 2026-09-08:
I'll go ahead and close this.

closed 2026-09-08 by GregValiant