Ultimaker/Cura #20322: Booting

Bretplays, 2025-02-26:
### Cura Version

5.9

### Operating System

Window's 11

### Printer

Anycubic Kobra Neo 2 

### Reproduction steps

All I tried to do was bout up the software.

### Actual results

When I try i wait 10 minutes. Nothing boots and it comes up as not responding.

### Expected results

It should boot

### Add your .zip and screenshots here ⬇️

I dont have any screenshots and I uninstalled it for the second time to reinstall it.

GregValiant, 2025-02-26:
Thanks for the report.
When Cura comes up, the USB Printing plugin looks for serial devices and attempts to communicate with them.  If you have USB devices plugged into the computer, and internally those devices are configured as serial devices, then USB Printing assumes they are printers.  An inordinate amount of time can be spent with USB Printing sending "M105", and your headphones going "What!!".

After Cura starts, go to the MarketPlace and click on the "gear" icon.  Scroll down and disable "USB Printing".
Let us know if that makes a difference.
There is a push to make USB Printing selectable from a menu command rather than being enabled by default.

Bretplays, 2025-02-26:
Well that's the thing. The slicer itself won't even boot up. It gets stuck at the like booting loading screen

GregValiant, 2025-02-26:
If it never comes up then I suspect some sort of local problem on your machine and I don't have any suggestions for that.  Maybe someone from the Cura team (@HellAholic ?) has an idea.

Bretplays, 2025-02-27:
Here's a screenshot of what I'm seeing. 

[image]

HellAholic, 2025-02-27:
Here are the general troubleshooting steps for resolving issues in Cura:

    Clear Cura's Data Folders
        Delete the contents of the following directories:
            %AppData%/Local/cura
            %AppData%/Roaming/cura
        This resets Cura to its default state, which can resolve plugin and configuration-related issues.

    Update System Drivers
        Ensure that all system drivers, including graphics drivers, are up to date.

HumbleDeer, 2025-03-11:
If this still doesn't resolve it, you can start Cura up with the extension disabled by going to the same cura folder in AppData as listed above, and adding its name to the list of disabled plugins in the appropriate json file (name escapes me right now)