Ultimaker/Cura #20326: Not able to select "Pressure Advance Control" under "Select settings" in Material preferences

DrWB63, 2025-02-27:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Creality Ender 3 S1 Pro with Sonic Pad (Klipper)

### Reproduction steps

1. Navigate to Preferences > Materials
2. Pick any material in dialog box
3. Go to Print settings tab in same dialog box, then open "Select settings"
4. Even with all "Show all" selected, the option to include "Pressure Advance Control" does not exist

### Actual results

The option to include "Pressure Advance Control" does not exist -- even though Pressure Advance Factor and Pressure Advance Smooth Time are available, which is inconsistent

### Expected results

I would expect that Pressure Advance Control would be available as a optional Materials setting.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-02-27:
Thanks for the report.
Those settings are added via the Klipper plugin from the MarketPlace.  You need to post this report on [their Git Page](https://github.com/jjgraphix/KlipperSettingsPlugin).
The developer of the plugin is responsible for maintaining it.  I'll remove the bug label and close this.  I hope you understand.

I installed the plugin and see this:
[image]

And in the 'settings visibility' dialog:
[image]

closed 2025-02-27 by GregValiant

DrWB63, 2025-02-28:
Thanks, Greg. Much appreciated.

 

I have just reported this issue following the link you suggested for the Klipper plugin.

 

Please note, however, that I see the same options as you do (re: the screenshots you posted) in the Klipper settings that are available with the plugin under the main print settings dialog box in the upper right of the Cura window. My issue is that the Pressure Advance Control setting is not visible if I follow these steps:

 

1.	In Cura, navigate to Preferences > Configure Cura > Materials
2.	Pick any material in dialog box
3.	Go to Print settings tab in same dialog box, then open "Select settings"

 

The option to include "Pressure Advance Control" does not exist, even with all "Show all" selected. Note that both Pressure Advance Factor and Pressure Advance Smooth Time are available, which is inconsistent.

 

I’m not sure if this a Klipper plugin issue or a Cura issue.

 

Are you able (now that you have installed the Klipper plugin) to replicate the issue following the steps above?

 

Cheers,

WB

 

image.png (view on web) <https://github.com/user-attachments/assets/db03b508-8b88-45b7-af96-bfdb05163a1a> 

image.png (view on web) <https://github.com/user-attachments/assets/dfa3dca2-865f-40dd-bfea-16874a8f1383> 

image.png (view on web) <https://github.com/user-attachments/assets/db03b508-8b88-45b7-af96-bfdb05163a1a> 

image.png (view on web) <https://github.com/user-attachments/assets/dfa3dca2-865f-40dd-bfea-16874a8f1383>

GregValiant, 2025-02-28:
I don't see any of the Klipper settings available "per material".  I had the "Material Settings" plugin installed prior to installing the Klipper plugin.  That might be why I don't have the settings to choose.

We are dealing with two plugins here and neither is supplied by UltiMaker.  How they are inter-relating and whether or not the loading order affects that, I have no idea.
In my case it seems reasonable that since the Material Settings plugin is loading first, and since at that point the Klipper settings don't exist, that they aren't available in my Manage Materials dialog.

UltiMaker Cura does not support Klipper as none of the UM printers use it.  It is left to outside contributors to add the functionality via plugins.  But it is at the very least a huge project (and maybe impossible) for a particular plugin author to go through every other Cura plugin in existence and determine if there are interferences.

DrWB63, 2025-03-01:
Greg – Thanks again! I have a growing appreciation for the complexity of this ecosystem. I had not even remembered that “Materials Settings” was a plugin. Oops. I will chase that down to as a possible source of the issue.

 

Cura is pretty cool piece of software and I’m glad I get to play with it. 😊

 

All the best,

WB

DrWB63, 2025-03-04:
Greg – I wanted to circle back and simply offer a note of thanks for your responses. As a result, you inspired me to dig in myself. I found that a single line of code was missing from the Klipper Settings plugin (in klipper_settings.def.json) that prevented the setting from being visible in the Material Setting plugin. Straightforward fix. I posted the solution on the plugin site. 😊

 

On a separate note, I have a different Cura question for you: I have a Creality Ender-3 S1 Pro (with Sonic Pad). When I look at the extruders folder in …/cura/5.9/ I see a single extruder cfg file with a “+%232” appended to the end of the filename. Within the file, I see “id = creality_base_extruder_0 #2” – what is the #2 in the filename and within the file referring to? Does Cura always do that or is this an artifact of prior definitions I’ve made?

 

Cheers,

WB

GregValiant, 2025-03-05:
I'm not the right person to ask about the config folder.  In doing the triage here I end up opening so many project files created by different people that I constantly have to delete printers and profiles.  As a consequence, my config folder "extruders" and "definitions" are a mess and those have to be cleaned out as well.  

The number "232" indexes upward whenever I open a project file involving another Creality printer.  Once in a while I'll get a message that some of the extruder files are corrupt (or something similar) and I have to go in and delete them.  The first two numbers "23" don't seem to have any significance and the last digit(s) is the one that indexes upward.
Below is what my config extruder folder is showing today.  You can see that I have 57 extruder 0's.  I have my Ender 3 Pro configured for up to 4 extruders and there are 49 extruder 1's, 49 extruder 2's, and 49 extruder 3's.  I know that is because I customized the Creality_base file and so any Creality project I open uses my custom base file and 4 extruders are added for each printer.

[image]

Picking an extruder file at random (it happened to be "creality_base_extruder_3+%2339.extruder.cfg"), I see this inside:
_id = creality_base_extruder_3 #39
machine = Creality CR-10S_

So yes, the number indexes upward whenever I open a project file involving a Creality printer.  I don't know why that is and quite frankly don't care as long as my printer works correctly and my machine settings don't get over-written by someone else's Ender 3 Pro.
Other than an all-metal hot end, my printer is stock and still runs the original Marlin 1.1.8 firmware.  I do keep up with maintenance and the prints are of excellent quality and dimensionally right on the money.