Ultimaker/Cura #20327: 【windows build from source code】

tzsh1007, 2025-02-28:
### Is your feature request related to a problem?

Currently,  use 'git checkout 5.10' switch  to this branch, and  exec 'conan install . --build=missing --update -g VirtualPythonEnv', two libraries has conflict between protobuf/5.27.0 and protobuf/3.21.12, the details in the graph

[image]

### Describe the solution you'd like

i don't know how to solve this problem

### Describe alternatives you've considered

 i [build by this tutor](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source#1-configure-conan)

### Affected users and/or printers

other people who wants to build by self

xenoscr, 2025-03-01:
Same issue from Ubuntu Linux:

```
conan graph info ... --format=html > graph.html
```

I've tried both main and the 5.9.1 release branches.

playwolf719, 2025-03-03:
Same issue

zebin-xlogic, 2025-04-07:
I have same issue. Did you solved it?