Ultimaker/Cura #20330: slicing Feheler

Benno1997, 2025-02-28:
### Cura Version

5.9.0

### Operating System

win 10

### Printer

flsun qqs-prio

### Describe model location

flach, mittig 

### Describe your model

nein 

### Add your .zip here ⬇️

[FQQS_tür spion v1 fehler.zip](https://github.com/user-attachments/files/19032438/FQQS_tur.spion.v1.fehler.zip)

GregValiant, 2025-03-01:
Thanks for the report.
The project would not slice for me either.  I checked the model with MS 3D Builder and it doesn't show any errors.
After moving the model on the build plate (it took a couple of tries) it did slice.
The second time I opened the project it sliced for me.
@HellAholic and the Cura team will take a look.

This is in 5.9.0
[image]

HellAholic, 2025-03-01:
[Repaired_FQQS_tür spion v1 fehler.zip](https://github.com/user-attachments/files/19038027/Repaired_FQQS_tur.spion.v1.fehler.zip)

Give the version above a try and see if it still crashes.

GregValiant, 2025-03-03:
That one sliced for me.
What did you use?  MS 3d Builder reported that there were no errors in the model.

HellAholic, 2025-03-04:
I used 3D builder but I'm on a AMD system with Win 11 so that might be the difference in detecting errors or not.
I'm looking into the model errors in free time, if possible doing a plugin that automatically shows up to repair the models.

GregValiant, 2025-03-04:
PrusaSlicer may have cut some sort of deal with AutoDesk to use NetFabb.  When it was available as an on-line service it was the best.  I think it's a "Windows only" app.

HellAholic, 2025-03-04:
This one specifically had the following errors based on my WiP code:
`Detected issues: {'flipped_normals': True, 'self_intersecting': True}`

I'm trying to see if I can get around it with open source options that are python based, makes the implementation much easier and not platform specific. Will probably drag you in for testing if I manage to make a somewhat working code 😆