Ultimaker/Cura #20333: Open File(s)... takes ages

Franticware, 2025-03-01:
### Cura Version

5.9.1

### Operating System

Manjaro Linux

### Printer

Creality Ender-3 v2 Neo

### Reproduction steps

1. started CURA
2. clicked File, Open File(s)
3. CURA seemingly did nothing for a while
4. after a few minutes, very simple model finally showed up

### Actual results

Opening stl model took too long.

### Expected results

Opening model should take seconds and show progress bar to indicate something is happening.

### Add your .zip and screenshots here ⬇️

[dell-soundbar.zip](https://github.com/user-attachments/files/19039040/dell-soundbar.zip)

GregValiant, 2025-03-01:
Thanks for the report.
Open Cura and go to the MarketPlace.  Click on the "gear" icon, scroll down to "USB Printing" and disable it.  After re-starting Cura see if the problem still exists.  Let us know either way.

If you have USB devices plugged into the computer, and they are internally configured as 'serial devices' (headphones, earbuds, etc.) then Cura will waste time trying to connect to them thinking that they are printers.

closed 2025-03-01 by Franticware

GregValiant, 2025-03-02:
USB printing is polling all the devices with several different baud rates trying to find one that works.  Eventually it gives up, but that takes a while.