Ultimaker/Cura #20336: Slicing error

Drastic396, 2025-03-01:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Ender V2

### Name abnormal settings

Standard setting

### Describe model location

no

### Describe your model

wing earrings

### Add your .zip here ⬇️

When trying to slice to a thickness of 1 or 1.2, the slicing fails.  I can slice to .8 or to1.4 or greater without issue.

[Angel Earing_22_pair_1.4mm.zip](https://github.com/user-attachments/files/19040497/Angel.Earing_22_pair_1.4mm.zip)

GregValiant, 2025-03-01:
Thanks for the report.
When I opened the file all the models were 1.4 thick and it sliced.
I made all the models 1.0 thick and they would not slice.
I grouped all the 1.0 models and rotated them 90° and it sliced again.
We have had some reports of these slicing errors lately.  I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2025-03-03:
Another case of model errors. Repair the model and it will slice without issues.

[image]

[image]