Ultimaker/Cura #20340: Full model renders; test portion missing rear fenders on car body

El-Vencedorr, 2025-03-02:
### Cura Version

5.9.1

### Operating System

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

Using Tinkercad I erased a portion of the model for a test print.  

Imported and sliced.

### Actual results

Fenders do exist as the tree or regular supports are still created for them in slice.  They show in slice preview as lightly-shaded and not solid.

Full-body model slice renders properly but not the partial test piece.  

Seems like a bug and not a feature.  :D 

### Expected results

Partial piece should render fenders along with the support structures that do become generated.

### Add your .zip and screenshots here ⬇️

[Bug Report.zip](https://github.com/user-attachments/files/19044399/Bug.Report.zip)

GregValiant, 2025-03-03:
Thanks for the report.
Those models are full of internal errors.  Without repairing them they won't slice or print.  TinkerCAD may have left open seams as well.  This is a 40 hour print and if it was me, I wouldn't take a chance on it
I'll remove the bug label from this and close it.  The models are so poor that they are most likely the problem.  I hope you understand.

You can see here that where the quarter panel should be is a clean cut surface.  That indicates an error in the model.
[image]

This is in PrusaSlicer.  There are a lot of areas missing because of random internal surfaces and open seams.
[image]

closed 2025-03-03 by GregValiant

El-Vencedorr, 2025-03-03:
Sorry to reopen with a reply.  The source for the model was the web and it
did report errors, however the whole body model renders.  Adding the clip
to it simply adds another level of confusion on top of the errors.  Thanks
for the prompt reply!

+ Kenn

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

GregValiant, 2025-03-03:
Before I closed the file in Cura I took another look at it.
It appears that someone took a shell of the model (might have been a scan) and then attempted to fill it with solid areas to make it printable.  You can see right into the "problem area" and it appears that TinkerCad did leave a large gap in the model.  When the slicer finds an area like that it can no longer figure out what is "inside" and what is "outside" and the slice fails, or is simply wrong.

I've marked the "real" model outer surface with the short red lines.  That shell is a surface only and has "0.00" thickness.  The black line approximates where someone added actual solid masses in an attempt to fill the model and make it printable.  They failed.
[image]

El-Vencedorr, 2025-03-03:
Thank you for the insight.  I appreciate the extra look-see.