Ultimaker/Cura #20343: build plate adhesion faulty with a circle shape

goofyseeker311, 2025-03-03:
### Cura Version

5.9.0

### Operating System

win 11

### Printer

elegoo nepture 3 pro

### Reproduction steps

1. design a thin circle flat shape (.stl).
2. load the model and slice with brim/raft.

### Actual results

brim/raft has gaps that should not be there.

### Expected results

the brim/raft should be full.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-03-03:
Thanks for the report.
That effect occurs when a model isn't flat on the bottom.  The brim/raft/skirt will only appear where the model actually touches the build plate.
If you look through the bottom of the build plate you will see that some of it is red (not in contact with the build plate) and other areas are blue (touching the build plate).

Without a project file I can't give a definitive answer, but this looks like a model problem.  You might be able to fix it by making the model 0.1mm taller in the Z and then setting the Z distance to -0.1 so it sits into the build plate.

The same effect can be seen with this model as only 1 little area is actually in contact with the build plate.

[image]

goofyseeker311, 2025-03-03:
the model stl file is flat so the slicing fails on recognizing that.

GregValiant, 2025-03-04:
This is why we request project files.  If this was a bug in the software thousands of people would be reporting it.

Post a project file as you were asked to when you filled out the original report form.

goofyseeker311, 2025-03-04:
added the .3mf files to the op. circle has this issue, square slicing works as expected.

GregValiant, 2025-03-04:
Here is the 3mf model sliced as received.
[image]

This is looking up from the bottom in the Prepare screen.  Only the cyan color is touching the build plate.
[image]

Then I use the "Lay Flat" tool.
[image]

And then slice again and the brim is complete.
[image]

goofyseeker311, 2025-03-04:
can you explain why your first slice is different than my first slice? slicing algorithm is not consistent. yep I manually angled the 45degree angled stl object to the surface, giving variable results. yep lay flat tool would solve it.

goofyseeker311, 2025-03-04:
your own solution to the problem shows proof that the slicing algorithm is not consistent and is broken.
by the fact that the object surface is flat. you will not get two different results if the algorithm is 100% correct.
assumption: model surface is flat.

GregValiant, 2025-03-04:
If you look at your first image closely, you will see this.
[image]

It is a step.  It is a different layer than the higher step at the top of the image, and the lower step at the bottom of the image.
There are two ways that can happen:

- The model is sitting flat on the Cura build surface and the top surface of the model is not planar-parallel to the bottom surface.
- The top and bottom surfaces of the model are in fact planar-parallel but the model is sitting at an angle on the Cura build plate.

Close inspection in MS 3D Builder reveals that the top and bottom of the ring are in fact planar-parallel and so the only conceivable way to achieve what we see in your images is that the model was sitting flat on the Cura build plate.  As I showed, when I lay the model flat on the Cura build plate there is no problem.

I'll remove the bug label and close this.  I hope you understand.

closed 2025-03-04 by GregValiant