Ultimaker/Cura #20351: Failed slicing when using tree support.

alparent, 2025-03-06:
### Cura Version

5.9.1

### Operating System

Windows 11

### Printer

Ender 3

### Name abnormal settings

Tree support

### Describe model location

no

### Describe your model

I sliced this yesterday, now I get this slicing failed

### Add your .zip here ⬇️

[AirWick_Kappe_v6.zip](https://github.com/user-attachments/files/19100269/AirWick_Kappe_v6.zip)

GregValiant, 2025-03-06:
Thanks for the report.
As received, it won't slice for me either.  One problem is a "Travel Speed" of 500 which is above the max speed in the machine settings.  Dropping that to 150 it still would not slice.
If I reset the custom profile "PLA - 0.20 5/3/2025" (and once again change the travel speed) it does slice.  That's with Normal supports.
If I change the supports to "Tree" slicing fails again.
I did not find any errors in the model.

@HellAholic of the Cura team will take a look.

HellAholic, 2025-03-06:
There are a couple of model errors present so when you force a slice the result looks like this 👇 
[image]

I did a repair on the model but still the crash happens with the profile settings in the 3mf so can't say for sure without a deeper investigation.
After repair it looks something like this with the default profile:

[image]

damian238, 2025-03-07:
For what it's worth, I d/led the 3mf and sliced as is without problem, changed to tree support and it still sliced without problem. 
I have an Ender5+ running Cura 5.9.0 on windows 10.

alparent, 2025-03-09:
There is definitely something going on with my profile. I also tried the standard profile and it sliced fine. IT never accrued to me to try another profile. Now to investigate what the issue might be? Any leads?

Thanks for your time.

GregValiant, 2025-03-09:
The problem occurs once in a while on certain systems.  It isn't a widespread problem (it's never happened on my computer) and so it's hard to get a handle on exactly what is going on.
Often, resetting a Custom Profile can allow the slice to proceed, but that isn't 100%.