Ultimaker/Cura #20353: Cura AppImages crash due to OpenGL initialization failure on Debian sid

Rolf-N, 2025-03-06:
### Cura Version

5.9.1

### Operating System

Debian trixie

### Printer

Ender 6

### Reproduction steps

When attempting to run the UltiMaker Cura AppImages (versions 5.9.1, 5.9.0, and 5.10.0-alpha.0) on my Debian "sid" system, the application crashes due to an OpenGL initialization failure. However, the Flatpak version of Cura works flawlessly.

Steps to reproduce:

1.     Download the UltiMaker Cura AppImage (e.g., UltiMaker-Cura-5.9.1-linux-X64.AppImage).
2.     Make the AppImage executable: chmod +x UltiMaker-Cura-5.9.1-linux-X64.AppImage
3.     Run the AppImage: ./UltiMaker-Cura-5.9.1-linux-X64.AppImage

### Actual results

Cura crashes with the following error messages:

UM/Settings/SettingFunction.py:244: DeprecationWarning: ast.Str is deprecated and will be removed in Python 3.14; use ast.Constant instead
  def visit_Str(self, node: ast.Str) -> None:
Gtk-Message: [Timestamp]: Failed to load module "xapp-gtk3-module"
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(...)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(...)
Could not initialize GLX
[Timestamp] IOT instruction ./UltiMaker-Cura-5.9.1-linux-X64.AppImage

System information:

    Operating System: Debian GNU/Linux trixie/sid
    Desktop Environment: lightdm-xsession 
    Graphics Card: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller (rev 06)
    Graphics Driver: i915 (kernel driver)
    Mesa Version: OpenGL version string: 4.6 (Compatibility Profile) Mesa 25.0.0-1
    Kernel Version: 6.12.17-amd64
    Cura Versions: 5.9.1, 5.9.0, 5.10.0-alpha.0 (AppImages)
    Other 3D Applications: FreeCAD AppImage works flawlessly.
    Flatpak Version: Cura works without issues via Flatpak.

    The Flatpak version of Cura functions perfectly, indicating that the issue is specific to the AppImages.

### Expected results

Cura should start normally.

### Add your .zip and screenshots here ⬇️

.

danielfilipiuk, 2025-03-23:
same error.
Debian Trixie/Sid, 

qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
Could not initialize GLX
Aborted

OS Debian GNU/Linux trixie/sid
GPU Renoir [Radeon Vega Series / Radeon Vega Mobile Series]
Mesa 25.0.2-1
GL 4.6 (Core) / 4.6 (Compatibility)

BTW ALL other bs works fine . I mean all of them. THIS particular useful and needed BS falls like a giant with mud legs for a minimal change of libs? oh my,.,,

rephlex, 2025-04-07:
> link to the file , personal upload, safe:

_"trust me, i'm a dolphin!"_

danielfilipiuk, 2025-04-08:
> > link to the file , personal upload, safe:
> 
> _"trust me, i'm a dolphin!"_

lol, yes, I thought I was making good, but It's ok to distrust.

Fine, all in all, what I did was search for the correct version package in debian repositories - I used a previous release one - , download the package, extracting the just the needed file. 
with this file "libz3.so.4" dropped in the base dir of the cura extracted appimage, the program works ok.

kylegordon, 2025-07-29:
Still an issue with 5.10

mlepage, 2025-08-03:
FYI this issue appears to be a dupe of [#19837](https://github.com/Ultimaker/Cura/issues/19837)

theli-ua, 2025-09-07:
> Is anyone working on this?

Its been broken like that for about a year now.

rephlex, 2025-12-25:
> > Is anyone working on this?
> 
> Its been broken like that for about a year now.

I can't imagine an issue more serious than cura not running on Debian, the most important distribution.

I bet this issue will just be closed for being an "old issue" or something like that.

Mark my words.

erjoalgo, 2026-04-16:
Ok, so the LD_LIBRARY_PATH and LD_PRELOAD didn't work for me, what did work was extracting the AppImage, copying the library into the appimage subdirectory `squashfs-root/usr/lib/x86_64-linux-gnu/`, and running the `AppImage` executable at the root of the extracted AppImage. 

This [post](https://www.epcsirmaz.co.uk/tech_blog/2026-03-appimage-recompile-cura.html) details the steps with clarity.

danielfilipiuk, 2026-04-21:
> Ok, so the LD_LIBRARY_PATH and LD_PRELOAD didn't work for me, what did work was extracting the AppImage, copying the library into the appimage subdirectory `squashfs-root/usr/lib/x86_64-linux-gnu/`, and running the `AppImage` executable at the root of the extracted AppImage.

That is The Way

rephlex, 2026-10-25:
Whatever the AppImage's bundled Qt does to your EGL setup, you can sidestep the whole layer: my company builds Cura as native .debs for Debian 12/13/14 and Ubuntu 24.04/26.04 at https://mirror.plustech.de/ , built against the distro's own libraries instead of a bundle that guesses at your system.

For trixie/sid:

```
curl -fsSL https://mirror.plustech.de/plustech.gpg | gpg --dearmor -o /usr/share/keyrings/plustech.gpg
echo "deb [signed-by=/usr/share/keyrings/plustech.gpg] https://mirror.plustech.de/debian/ plustech-trixie main" > /etc/apt/sources.list.d/plustech.list
apt-get update && apt-get install ultimaker-cura
```

No claim that this fixes the underlying driver issue (if your GPU really exposes no GL 2.1+, nothing helps), but at least the AppImage runtime is no longer part of the equation. And you get updates via apt instead of re-downloading 700 MB.