Ultimaker/Cura #20354: Add Printer Screen can't be moved or sized

goulardi, 2025-03-06:
### Cura Version

5.9.1

### Operating System

Windows 10

### Printer

Couldn't select any

### Reproduction steps

Went to add a Creality Ender 3 Max Neo (which I don't think is on the list) printer and the window that pops up can't be moved or resized and the lower portion is cut off from the screen.  

### Actual results

Nothing to show.  See screenshot below.

### Expected results

No comment.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-03-07:
Thanks for the report.
I think the "Networked" printers are for UltiMaker printers only.  If you look at the "Non-Networked" printers there is an extensive list of Creality machines.  There is an Ender 3 Max and an Ender 3 V2 NEO, but no Max NEO.
The dialog may not be re-sizing because there isn't anything to show on that page.

Creality has stopped submitting printer definitions to UltiMaker.  I suppose that is due to their wish for people to use Creality Slicer (or whatever it's called this month).

Try adding a "Non-UltiMaker" printer and then "Non-networked" printer and see if that works for you.

Right now I don't see a bug here so I'll remove the label (at least for now).

fieldOfView, 2025-03-07:
@GregValiant, you are missing the point of the issue; the "Non-networked printer" button is below the visible part of the window in the provided screenshot. OP can not see that button.

Workarounds for OP:
* Try if pressing the "Add a networked printer" heading collapses that section
* Set the taskbar to autohide
* Set a lower DPI setting in Windows

GregValiant, 2025-03-07:
Ah...  And the dialog cannot be made shorter, only taller.

goulardi, 2025-03-07:
I have very limited programming experience, but don't most languages that
include "windows" have a switch setting (during programming) to make a
window to include a minimize, maximize, & close buttons as well as giving
the window the ability to be (mouse) "dragged"?  I only brought this up
thinking this might be addressed in future revs.  Thanks.