Ultimaker/Cura #20357: cura 5.9.0 slicing failed with change to support z overide.

damian238, 2025-03-07:
### Cura Version

5.9.0

### Operating System

windows 10

### Printer

Ender5+

### Name abnormal settings

slicing failed after setting "z overrides x/y" resetting back to "x/y overrides z" was slicable.

### Describe model location

not tested

### Describe your model

It was and is still slicable, just not with the override  set to prioritize z.

### Add your .zip here ⬇️

[walker.zip](https://github.com/user-attachments/files/19127332/walker.zip)
This may not be this best place to do so, but I want to express my heartfelt thanks to all the people at Ultimaker. You truly help making a better world more possible.

GregValiant, 2025-03-07:
Thanks for the report.
The part is not really in contact with the build surface.  You can see here that there is only a little spot of "cyan" indicating the contact patch.
[image]

This is the best that the "lay flat" tool could do.  The bottom of the model isn't flat, and I have a feeling there are some other issues in there.
[image]

This slice is after I settled the model on the build plate.  It just doesn't look right to me.  The wavy lines here and there seem to indicate model problems.

GregValiant, 2026-09-08:
I'll go ahead and close this.

closed 2026-09-08 by GregValiant