Ultimaker/Cura #20360: skirt goes through model after slicing

sidmeee, 2025-03-07:
### Cura Version

5.9.0

### Operating System

ubuntu mate 24.04.2 LTS (Noble Numbat) 64-bit

### Printer

ender 3 

### Reproduction steps

when i sliced a model i added a skirt to it and when it slices it looks good in the preview  with a 5 line count at 4mm from model 

[image]

### Actual results

after saving i reloaded it and the skirt is broken and through my model 

[image]

### Expected results

[image]

### Add your .zip and screenshots here ⬇️

[image]

[image]

closed 2025-03-07 by sidmeee

GregValiant, 2025-03-07:
Thanks for the report.
You found it eh?
Do you happen to have the "Arc Welder" plugin active?  Cura's preview doesn't understand G2 and G3 moves and so radii are missing and the straight lines connect erroneous line ends.  That's why you see errors on the sides of the model as well.

sidmeee, 2025-03-07:
Hi  yes i did .but it printed exactly like the messed up preview . this has been going on for a while but i didnt notice it until i started printing curved models , really confused me for a while .. i deleted arc welder and its back to normal , thank you

GregValiant, 2025-03-07:
The stock firmware on early Ender models supported G2 and G3 moves so Arc Welder didn't affect the printed output, just the preview of a gcode.
On the new models I don't have information on the firmware.  If the prints were coming out like the preview then my guess would be that G2/G3 are not supported by your firmware.

sidmeee, 2025-03-08:
the funny thing is when i reload the saved gcode back  into cura it looks like the messed up print exactly, i think its a software issue on the save function because when i look at the preview its fine so i saved it and printed it and its messed up , maby something in arc welder is not right

GregValiant, 2025-03-08:
Arc Welder has been around for years now and I can't recall an instance where it was a problem.  I think the firmware just isn't configured for G2/G3 moves.
The preview you see right after a slice is without Arc Welder.  It runs after the gcode is produced and during a save and so the gcode file has G2/G3 moves, but the original preview does not.
Then when you open the Gcode file in Cura, the gcode reader sees the G2/G3 moves and ignores them.  That's what the printer is doing as well.  Instead of an arc between two points you get a straight line.

sidmeee, 2025-03-08:
that makes a lot of sense thanks for your help it cleared up a lot for me       

thank you