Ultimaker/Cura #20362: issue

AnhDuy852003, 2025-03-08:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

Anet E12

### Describe your model

yes

### Add your .zip here ⬇️

[AE12_nap.zip](https://github.com/user-attachments/files/19139959/AE12_nap.zip)

HellAholic, 2025-03-08:
Apart from the support not matching (one of them had 0 length in x & y), model had errors.
Slices after repair, and I mirrored the Supports on the right side to the left side, you can adjust if you want.
[image]

Repaired project.

[fix_repair_nap.zip](https://github.com/user-attachments/files/19144578/fix_repair_nap.zip)

Probably gonna remove/replace the text for the slice fail to point towards how to repair the models rather than report on github as more than 90% so far has been model issues.

closed 2025-03-08 by HellAholic

GregValiant, 2025-03-09:
This is another one.
I checked the model with MS 3D Builder and it showed no errors.  Checking with "https://formware.co/OnlineStlRepair" it didn't show any errors but did make adjustments to the triangle and vertex counts.  I also used the NetFabb repair in PrusaSlicer.
I couldn't get any of those "repaired files" to slice.

My own non-scientific count is:  ~90% model errors, %5 user errors (bad setting combos and such) and about 5% Cura problems.

HellAholic, 2025-03-09:
I have a spike (investigation ticket) planned to look into the model errors and how it should be handled in Cura.
First need to figure out what part in the model geometry is causing the issue, then if that is consistent and repeatable then we can have a detection method in place.

Main thing is that at the moment we're categorizing all of the errors/issues under the slice fail error and that is not accurate. This does not guide the users towards fixing issues but rather pointing them to one location for collecting all errors and at the same time the fix might take them 1 or 2 minutes instead.

Example of this model, I can remove the errors with my script so that the 3D builder does not detect any model errors, but still it will not slice, so TLDR:
under investigation