Ultimaker/Cura #20363: Slicing failed

fiftystiff, 2025-03-08:
### Cura Version

5.9.1

### Operating System

Windows 10

### Printer

Ender 3

### Name abnormal settings

Default, completely fresh install

### Describe model location

No

### Describe your model

Rectangular test box, but no models slice

### Add your .zip here ⬇️

[Box.zip](https://github.com/user-attachments/files/19141519/Box.zip)

I have tried installing multiple versions and wiping every related file within AppData when uninstalling to do a completely clean install; nothing slices at all. I used to have no issues slicing and the first time slicing failed was right after I added a CR-10S. Since then slicing has never worked even after uninstalling/reinstalling multiple different versions.

GregValiant, 2025-03-08:
Thanks for the report.
The problem appears to be system specific.  I didn't have any trouble slicing in 5.9.0 with a CR-10S active.  (This is on a Windows 10 Pro laptop.)
After a failed slice, use the "Help | Export file for Technical Support" command and then go to the file, zip it, and post the zip folder.  It will have the cura.log file in it.  Maybe it will give a clue.

fiftystiff, 2025-03-08:
[20x20x20 Cube Export.zip](https://github.com/user-attachments/files/19145763/20x20x20.Cube.Export.zip)

File attached above is generated from the same machine as before, installed Cura 5.9.1; added CR-10S as printer, didn't change any print settings, slicing failed. I made a 20x20x20mm cube in Tinkercad to basically rule out the model as the problem.

I also just tried on another windows 10 PC that never has had any version of Cura installed; I downloaded/installed 5.9.1, launched, clicked through the setup stuff (without signing into Ultimaker account), added CR-10S when prompted, didn't change any settings or anything, drag/dropped file into Cura, and slicing failed.

fiftystiff, 2025-03-08:
Another note to add: changing printer types still does not slice. I have an Ender 3 and a CR-10S, never had any issues until I added the CR-10S recently, but even when switching the printer to the Ender 3 slicing still always fails (no matter what model, including models that have previously sliced without issue).

GregValiant, 2025-03-08:
I'm not having any trouble slicing it.  I have about 20 printers installed of all sorts and checking with 5 or 6 they all slice the model.
We could use that log file.

fiftystiff, 2025-03-08:
I attached in my reply, is that the right thing? It was generated using the "Help | Export file for Technical Support" command you mentioned.

GregValiant, 2025-03-08:
That's my bad.  I opened it as a project.
From the log file is this line indicating a crash:
"Backend exited abnormally with return code 3221226505!"

I'm not familiar with that code.  Someone from the Cura team will take a look.

fiftystiff, 2025-03-09:
I figured it out; I looked through the logs and it looks like the "Sentry" part was failing (see log below):

[MainThread] UM.Backend.Backend.startEngine [96]: Started engine process: C:\Program Files\UltiMaker Cura 5.9.1\CuraEngine.exe
[MainThread] UM.Backend.Backend._backendLog [140]: [Backend] Calling engine with: ['C:\\Program Files\\UltiMaker Cura 5.9.1\\CuraEngine.exe', 'connect', '127.0.0.1:49675', '']
[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-03-07 20:40:45.224] [info] Sentry url: https://734f9ec9024f73e53701d59c3ffddfe3@o323038.ingest.sentry.io/4506257745510401
[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-03-07 20:40:45.224] [info] Sentry config path: C:/*******s/*******/AppData/Roaming/cura/.sentry-native
[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-03-07 20:40:45.224] [info] Starting sentry
[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-03-07 20:40:45.236] [info] Connecting to 127.0.0.1:49675
[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-03-07 20:40:45.237] [error] Could not connect to the given address
[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-03-07 20:40:45.350] [info] Closing sentry
[EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-03-07 20:40:45.443] [info] Closing connection.
[MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.9.1\share\uranium\resources\shaders\toolhandle.shader]...
[MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [1065]: Backend exited abnormally with return code 3221226505!

I disabled my VPN and it worked immediately, then re-enabled the VPN and slicing failed; then I disconnected my network cable and slicing succeeded, so it's probably a bug with the VPN not allowing connection to "Sentry" and Cura getting hung up on that rather than just recognizing that it can't connect.

fiftystiff, 2025-03-09:
I also checked on my VPN settings; I have split tunnel enabled and Cura is not on the list of programs that should be running through the VPN, so I am not sure if that is a bug with split tunnel with my VPN or a bug with Cura.

GregValiant, 2025-03-09:
You can try going to "Preferences" and the "General" tab and scrolling to the bottom.  Turn off the "Privacy" and "Update" options.  With those off, I would think that Cura wouldn't try to access a network.