Ultimaker/Cura #20375: Slicing failed for very basic model

kcl93, 2025-03-12:
### Cura Version

5.9.1

### Operating System

Windows 10

### Printer

Anycubic i3 Mega

### Name abnormal settings

Normal settings print. I have many custom settings but it seems like setting the infill back from gyroid to zig zag fixes the issue.

### Describe model location

center and the flat site down

### Describe your model

No. It is watertight and fixes with the Mesh Fix plugin do not help

### Add your .zip here ⬇️

[AI3M_Wandhalterung.zip](https://github.com/user-attachments/files/19206399/AI3M_Wandhalterung.zip)

HellAholic, 2025-03-12:
[Repair_AI3M_Wandhalterung.zip](https://github.com/user-attachments/files/19206612/Repair_AI3M_Wandhalterung.zip)

The model you're using has errors, you can fix it with a 3D software like 3D builder on windows, it's free and it works nicely.
There is a repaired version of your project above, you can give that one a try. It should slice without issues.

kcl93, 2025-03-13:
Hi @HellAholic 
Thanks for the very quick answer and sending a fixed model. Are you able to elaborate what errors it contained? The Mesh Fix plugin in Cura did not indicate any issues.
Did you fix it with this tool? https://3d-builder.en.softonic.com/

HellAholic, 2025-03-13:
I used Microsoft 3D Builder which is a free Windows app and you can get it through the Microsoft store instead of the link you shared (that way you will get the updates as well and it's a trusted source).
Even with the 3D builder on different systems the same model would show/not show errors so that's why we have an investigation ticket into this. I'm using your model as one of the test cases to compare, once I know more I can share it with you, but at the moment I just used a tool to repair the model and it didn't generate any reports.

kcl93, 2025-03-13:
Hi @HellAholic 
I am a bit confused. 3D Builder does not seem to be available for download anymore in the Microsoft store. I tried the version from Softonic (v20.0.4.0) but it is not detecting any issue whatsoever with the model. Converting it between .3mf and .stl using the 3D Builder yielded the same unspecific error message. What kind of errors did the 3D Builder claim to find in your case?

HellAholic, 2025-03-14:
My guess is that it optimizes the model for slicing, the main change I see is the change to the number of polygons visually. As I mentioned, it doesn't show any error but rather a fix model button that appears when you load the model. There is also no option to get that information from the software. It is also system specific so me and GregValiant have been discussing this as the same model that shows and error on my device does not show any issues on his device.

You can load both the original model and the one after repair and do a compare between them to see what changed.

kcl93, 2025-03-16:
@HellAholic thanks for your explanations! I will attempt to reduce the numbers of faces when exporting in the future if I come around the same issue again.

closed 2025-03-16 by kcl93

HellAholic, 2025-03-17:
Small update:
One of the devs took a look at the slice issue with your model. With a general fix, the original model is also slicing without errors (error is being handled by the engine).
The fix should be part of the 5.10 Stable.
Left (5.10 Beta - slice error) - Right (build with the fix)
[image]