Ultimaker/Cura #20376: 5.10.0-beta.1 slicing error (went well in former version)

EmEiBee, 2025-03-12:
### Cura Version

5.10.0-beta.1

### Operating System

Win10 PRO, GTX1050Ti, Ryzen 7 1700, 64gb RAM

### Printer

I3 clone (DIY) via Repetier

### Name abnormal settings

Using CURA default "Normal"

### Describe model location

No. Placement independent

### Describe your model

Sliced (and print) without problems yesterday with former CURA version; ocures only in this BETA

### Add your .zip here ⬇️

[Aufsatzkuhler_1v0.zip](https://github.com/user-attachments/files/19207288/Aufsatzkuhler_1v0.zip)

HellAholic, 2025-03-12:
Hey @EmEiBee ,
Thanks for the report. The project file slices without issues on the latest version as far as I can see. Could you please provide a technical support package which also includes the Cura logs?

You can get to it via the Help menu on top bar > Export Package For Technical Support
[image]

EmEiBee, 2025-03-12:
... sure, I can ;)

Here it is...

Greetz
Micha

[export-20250312-115320.zip](https://github.com/user-attachments/files/19208461/export-20250312-115320.zip)

HellAholic, 2025-03-12:
In your printer definition you need to add a line to inherit from fdmprinter or add the new settings manually to your custom definition.
```
    "version": xx,
    "name": "your printer",
    "inherits": "fdmprinter",
```

Since you have a DYI printer, I think you have a custom printer definition and that one is most likely missing the inherit bit as the log indicates a missing setting value. This setting in addition to few others has been updated in 5.10
`[error] Trying to retrieve setting with no value given: wall_overhang_speed_factor`

EmEiBee, 2025-03-12:
> In your printer definition you need to add a line 

Where or in which file do I have to add this and where can I find it?

Greetz
Micha

HellAholic, 2025-03-12:
Without knowing how you added the printer (manually, through a package, etc.) or checking your configuration folder, I can't pinpoint the exact location.

However, you can check the configuration folder (Help > Show Configuration Folder or %appdata%\Roaming\cura\5.10\). If there are files in the definitions folder, the one matching your printer's name likely needs updating. In that file you should add 
`"inherits": "fdmprinter",`.

Before modifying any files, ensure Cura is closed.

The most probable issue is an outdated fdmprinter.def.json file in the configuration folder.

If you find fdmprinter.def.json there, try removing it and check if the printer still works. Alternatively, you can update it from the Cura installation directory:
Share > Cura > Resources > Definitions > fdmprinter.def.json.

That said, this is not the correct way to add a printer to Cura, as it leads to issues like this. Printer files should directly reference Cura's built-in resources.

EmEiBee, 2025-03-14:
In the meantime, I tried replacing the missing entry. But unfortunately, that didn't solve the problem.
I then completely deleted the previous version and the BETA, along with the registry entries, and then reinstalled the BETA.

That helped. It works now.

So, it seems that fragments of the previous version (5.9.1) were somehow interfering with the BETA; whatever...

closed 2025-03-14 by EmEiBee

HellAholic, 2025-03-14:
Glad you found a workaround and things are in working order. I'll add an investigation ticket just to be safe. If you add the printer with the internal flow, it should not have caused that issue.

EmEiBee, 2025-03-15:
> If you add the printer with the internal flow, it should not have caused that issue.

That's not true. As I already mentioned, I added the printer WITH THE INTERNAL FLOW.

I think there are deficiencies in the update routine here. I had a similar problem 2 or 3 versions ago, where nothing worked after an update. Back then, the only thing that helped was a complete uninstall and reinstall.

There are still a number of issues in CURA that haven't been addressed yet, such as the output of helpful error messages, for example, for a syntactical error in the start/end Gcode (comma instead of a period, unknown commands, etc.), or the cumbersome handling of different nozzle sizes on the same printer/hotend, etc.

But that's okay. You can't make everything foolproof.

HellAholic, 2025-03-15:
The focus was on 'it should not break in that flow so something is potentially going wrong in the upgrade flow' in that sentence rather than something you should have done in another way.

EmEiBee, 2025-03-15:
I have no idea what you're talking about, or rather, I don't understand what you're trying to tell me. Probably a language issue.

But it doesn't matter. I've learned from this that it's best to completely uninstall the old version before installing a newer one. That seems to be the safest approach, at least for my software environment. All good...