Ultimaker/Cura #20382: error slicing

johnbiz247, 2025-03-13:
### Cura Version

5.9.0 beta 2

### Operating System

windows 11

### Printer

elegoo neptune 3 plus

### Reproduction steps

just tried to slice

### Actual results

error and slices after 3rd try

### Expected results

slice

### Add your .zip and screenshots here ⬇️

28115

johnbiz247, 2025-03-13:
none

MariMakes, 2025-03-14:
Welcome to the UltiMaker Cura Github 🚀 

Have you tried reverting to the default settings?
Have you tried slicing with Support disabled?
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: 
https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

johnbiz247, 2025-03-14:
Yes i have, i dont get all the time. Never got on 5.9.0 beta1 and below. I
check all files before slicing. Like i mention it errors once or twice then
it slices. Just a glich maybe in sometime in 5.9.0 beta2 not in bata1.