Ultimaker/Cura #20391: Supported part first layer speed / use bridge settings

sargonphin, 2025-03-14:
### Is your feature request related to a problem?

When printing supported parts, it is preferable to have a small vertical spacing (Z distance) between support roof and supported part to make removal easier. Because of that spacing, it is necessary to print the first layer of a curved supported object very slow to give time for the filament to drop and solidify

Currently, Cura does not provide such a feature and thus, supported walls and skin are not printed correctly. The walls are printed at regular walls speeds, which is too fast and the filament "drags" in a straight line instead of a curve (the pictures below show a wall that was printed at 180 and 240mm/s over a support roof, with supports already removed)

[image]
[image]

### Describe the solution you'd like

2 possibilities, one preferable to the other but the other is already implemented:

- A new set of settings for the first layer over supports, with the possibility to expand these settings to 3 or more layers instead of just 1 (useful when printing with small layers heights) (preferable solution)
- Layers directly over supports use the same speeds as Bridge settings

### Describe alternatives you've considered

A setting exists to print an unsupported skin layer using Bridge settings (Bridge Skin Support Threshold in Experimental) which allows skin to be printed over supports more easily, but walls do not use these settings/don't have an equivalent setting and thus are printed too fast

### Affected users and/or printers

Only users using fast printers that print fast walls will benefit this change

HellAholic, 2025-03-14:
@sargonphin Feature is part of the 5.10 Beta release, you can download that version and try it out.
Search for Bottom Surface and all the settings related to that should pop up.

https://github.com/Ultimaker/Cura/releases/tag/5.10.0-beta.1

[image]

sargonphin, 2025-03-16:
Thanks for your reply, but this does not resolve the problem I am having here. These settings are for bottom layers only, but in my case the walls that need to be printed slow are not at the bottom. After downloading the new version and setting up the features, the wall speeds did not change

[image]

Here are a couple screenshot to show what part I am trying to print

[image]
[image]

Let me know if there are more settings I did not see that should be changed. Here is the file for the part in question

[TubeDuct_Pump.zip](https://github.com/user-attachments/files/19273140/TubeDuct_Pump.zip) (3mf file inside)

HellAholic, 2025-03-16:
How many "bottom surface skin layers" have you set in the profile if the number is 0 then the settings will not apply.
With 2 layers & setting bottom surface outer/inner wall speed to 25 I get this which includes the walls.

[image]

pkuiper-ultimaker, 2025-03-16:
> internal reference @Ultimaker/cura
>    - All settings for the bottom surface skin should have enabled if
>    'Bottom Surface Skin layers' > 0

We use 1 layer in the profiles.

sargonphin, 2025-03-16:
Thanks for the fast reply! @pkuiper-ultimaker I am not using any default profiles, as this printer is custom-built. The profile is also custom

@HellAholic Thank you for the tip! I had 0 bottom layers in my settings. I am encountering an extra problem though, where the bottom layer is only partially printing slow. This leads to very sharp changes in flow which even with the direct extruder would show on the surface

[image]

I will search for a solution in the background, if I find anything I will get back to you. Thanks again for the help, very appreciated :)

pkuiper-ultimaker, 2025-03-16:
You can set the speeds for the bottom surface (floor) outerwall, inner wall
and skin. You have to set them all the resude the speed on the floor
completly.

Also be aware that if the floor layer is not supported it will be
considered a bridge and the bridge settings are applied (if you turned them
on).

closed 2025-03-20 by sargonphin