Ultimaker/Cura #20392: .STL file locked (owned?) to Cura

keith20mm, 2025-03-14:
### Cura Version

5.9

### Operating System

Windows 10

### Printer

Not related

### Reproduction steps

I generate a .stl file using Alibre or Cubify.

I cannot delete the file from Explorer, nor from within Cubify file manager.. it is somehow owned by Cura.

From within Cura, I can "Load" using the explorer, but not load, just delete the .stl file, no problem  It appears the stl extension becomes owned by Cura.

### Actual results

File operations on .stl outside of Cura are prohibited.

### Expected results

The file should simply be renamed, or deleted, using file explorer, or from within Alibre or Cubify.

### Add your .zip and screenshots here ⬇️

This problem has nothing to do with any model, only with the file extension .stl

GregValiant, 2025-03-15:
Does this report have to do with the "Default Program" to open STL files?  When you right-click on a file in Windows File Explorer what does the top line in the menu say?

[image]

If you go to "Properties" and "Opens With" and "Change" can you change to the app of your choice?

keith20mm, 2025-03-17:
Well, .stl was actually opened by pythonW.  Why?  I have no idea.  It was a couple of years back when pythonW was installed.

Strange thing is that double-clicking a .stl file in Explorer started Cura 15.x.x, an ancient version, and I currently use the most recent release Cura.

So, I changed the extension association to UltraEdit, a text/hex editor.

No more problem.  I am closing this dog.

closed 2025-03-17 by keith20mm

GregValiant, 2025-03-18:
Well you've impressed me with your detective skills.  Nice job chasing that down.