Ultimaker/Cura #20394: Slicing failed :(

RobotoconYT, 2025-03-14:
### Cura Version

5.6.0

### Operating System

Windows 11, RTX 4060 ti, Ryzen 7 8700g

### Printer

Ender 3 S1 Pro

### Name abnormal settings

None. Default settings.

### Describe model location

I changed the size from 50% to 85%

### Describe your model

Dummy 13 Skeleton

### Add your .zip here ⬇️

it was a .3mf file, so i can't add it

Asterchades, 2025-03-15:
The first thing you should do when encountering a slicing issue is to update your Cura installation. Every minor build of Cura has included additional fixes with regards to models failing to slice, with a fairly major fix implemented with 5.7.0. The current latest version as of writing is 5.9.1, with 5.10 in beta.

> Please make sure to .zip your project file.

This is mentioned when creating your issues. GitHub imposes its own file type limitations that cannot be altered - you either need to compress it as asked or manually change the extension to an allowed one.

RobotoconYT, 2025-03-17:
I figured it out, that was it. I had used a 3mf file that was converted to stl, that was the issue.

closed 2025-03-17 by RobotoconYT