Ultimaker/Cura #20396: Cura signing me out

kweeperdev, 2025-03-15:
### Cura Version

5.9.1

### Operating System

Windows 11 pro

### Printer

Creality Ender 3 (modifieed to work for Voxelab Aquila)

### Reproduction steps

Open Cura
Sign in
Enter details in browser window
go back to cura
wait like 10 seconds

### Actual results

I get signed out and when i go to sign back in the browser window just says this:

UltimakerAccount
Not signed in
Please allow the application to access the requested data. You can do this by signing in again and clicking Allow.

### Expected results

It signing me in and staying signed in

### Add your .zip and screenshots here ⬇️

[image]
[click sign in button]

[image]
[signed in]

[image]
[did not sign into cura]

[waiting for like 30 seconds]

[blank page saying that i need to click allow and sign in again]

GregValiant, 2025-03-15:
Thanks for the report.
I'm curious why you need to sign in to an UltiMaker account.  This report says you have a modified Creality printer rather than an UltiMaker printer.
I've never had to sign in for anything for my Ender and I believe that any network support/connectivity would be for UM printers only.

kweeperdev, 2025-03-17:
I moved to a new desktop and I just wanted to sync my plugins and stuff. (It has nothing to do with the printer, I just filled that out as it was part of the form)

GregValiant, 2025-03-17:
The syncing should happen upon install.  Cura checks the earlier configuration folder and moves things up.  I just went from 5.9.1 to 5.10 beta and everything moved over.
Some of the plugins won't work if the author has a "version" check in it.  You would need to wait for those to be updated.
Most of FieldOfView's plugins have a version checker.  He's real good about keeping up and he's conscientious.  He also has several plugins to go through and check so it may be a bit.