Ultimaker/Cura #20397: Adaptive Layer height minimum wrongly calculated with <0.1 values

wolfsilver00, 2025-03-15:
### Cura Version

5.9.1

### Operating System

Archlinux

### Printer

Creality mk10

### Reproduction steps

Good known configuration (both with 0.2 layer height):

[image]

What happens if you go lower than 0.1:

[image]

### Actual results

As you can see, in the first image, it calculates correctly 0.2 - 0.1 = 0.1 for minimum, but on the second image, it substracts an additional .1 from the minimum value (does not seem to happen to the maximum under any circumstance that ive tried so far)

### Expected results

First image is correct, second image range should be 0.12 to 0.28

### Add your .zip and screenshots here ⬇️

Added screenshots in reproduction steps, as other configs do not seem to matter

GregValiant, 2025-03-15:
Thanks for the report.
This looks like expected behavior to me.

In your first instance:  Your layer thickness is 0.20 and your Maximum Variation is 0.1.
Minimum Layer Height = (0.20 - 0.10) = **0.10mm**
Maximum Layer Height = (0.20 + 0.10 = **0.30mm**

In the second instance:  Your layer thickness is 0.10 and your Maximum Variation is 0.08
Minimum Layer Height = (0.10 - 0.08) = **0.02mm**
Maximum Layer Height = (0.10 + 0.08 = **0.18mm**

Since you did not include a project file I have no way to check - but in that second instance did you happen to have the "Initial Layer Height" set to 0.28?  That would make the maximum layer height in the file 0.28 rather than a maximum height of 0.18.

I'll remove the bug label and close this as Expected Behavior.  Your statement _"... second image range should be 0.12 to 0.28..."_ would seem to be in error.  How could the computed minimum be above the set layer height?
Let me know if you feel that is premature.

closed 2025-03-15 by GregValiant

wolfsilver00, 2025-03-15:
[image]

Adding a screenshot with settings filtered so that you can see how everything is set-up for more clarity

wolfsilver00, 2025-03-15:
[image]

<3

GregValiant, 2025-03-15:
You we're asked for a project file when you filled out the report.  Please post one as I cannot duplicate this problem.  

[image]

wolfsilver00, 2025-03-15:
> You we're asked for a project file when you filled out the report. Please post one as I cannot duplicate this problem.
> 
> [image]

on it

wolfsilver00, 2025-03-15:
[PK3_dinoBlob1.zip](https://github.com/user-attachments/files/19266503/PK3_dinoBlob1.zip)

wolfsilver00, 2025-03-16:
@GregValiant 

[image]

If it helps, opening the same project file on cura 5.9.0. without modifying anything, gives me what is expected

GregValiant, 2025-03-16:
I still can't duplicate this.
Because you have an "Unknown" custom machine, it doesn't load when I open the project.  Consequently I am still working with my printer active.  What are you using for a printer definition file?  Does it have an "Adaptive" quality file?
This is with .2 layer height and .08 max variation for adaptive.

[image]