Ultimaker/Cura #20399: Arti_Snake_180_update.stl 70 MB  |  February 11, 2025

Gixxer0001, 2025-03-16:
### Cura Version

5.9.1

### Operating System

windows 11

### Printer

ender 3 pro

### Name abnormal settings

 None to speak of

### Describe model location

center of build plate, flat

### Describe your model

Arti_Snake_180_update.stl 70 MB  |  February 11, 2025

### Add your .zip here ⬇️

it just will not slice no matter what I try

Gixxer0001, 2025-03-16:
will not slice no matter what I do

GregValiant, 2025-03-16:
When you filled out the bug report you were asked to provide a project file and your Cura log file.  Without those there isn't anything that can be done.

Gixxer0001, 2025-03-16:
I don’t know what I did but somehow I got it to work. I didn’t save it to a
disk. I saved it to a file instead maybe that helps. I don’t know I’m new
with this. Sorry about that anyway but thanks for you guys. Help

closed 2025-03-16 by Gixxer0001

Gixxer0001, 2025-03-16:
I’m also having troubles with the program not responding is my computer not
good enough?

GregValiant, 2025-03-16:
From what I have seen around here, problems are generally with the way Cura interacts with your hardware.  There have been problems with "multiple monitor" systems, and with some video sub-systems, that sort of thing.
I run Cura on an older laptop that I got "used".  It's been fine running on Windows 10 Pro.  This laptop is no great speed demon and is probably 5 years old.

There have been some reported problems when USB devices are plugged into the computer.  Some of them might be configured as "serial devices" and Cura thinks they are printers and keeps attempting to "handshake".  Things like earphone buds don't respond and so the program can get confused.  There have also been some conflicts with other software but that seems to be rare.

Gixxer0001, 2025-03-16:
Thanks for the feedback! Very helpful!!