Ultimaker/Cura #20404: 'Extensive Support Logging' plugin seems broken in 5.10 beta

GregValiant, 2025-03-17:
### Cura Version

5.10 beta1

### Operating System

Windows 10 Pro

### Printer

Ender 3 Pro

### Reproduction steps

Upon installing 5.10 beta 1 I was informed that "Extensive Support Logging" would need to be un-installed.
I looked through it a bit looking for a 'version blocker' but I didn't find one.
Re-installing it didn't work.

### Actual results

Had to uninstall the plugin.

### Expected results

It works as it did.

### Add your .zip and screenshots here ⬇️

This is just a question, but what is this spaceship looking thing?  It should say "File"?  Occasionally a tooltip will peek out that says "Favorites".

[image]

MariMakes, 2025-03-18:
Hi @GregValiant,

Thanks for the report! 💟 
We've added a ticket with the intent to improve this for stable. 
For internal reference CURA-12476

GregValiant, 2025-03-25:
@MariMakes Both of these issues are still problems in 5.10 stable.

rburema, 2025-03-26:
@GregValiant -- We're releasing a new version of the plugin for 5.10, we have to, since we upgraded to python 3.12

MariMakes, 2025-03-27:
We updated it on the marketplace so it should be working again!
Thanks ofr the report!

[image]