Ultimaker/Cura #20408: Provide a means to easily compare any two profiles.

mbmast, 2025-03-18:
### Is your feature request related to a problem?

The problem is that comparing any two profiles involves exporting each of them and then importing them into another tool, like WinDiff or maybe Edit+ to compare the differences between the two profiles.

### Describe the solution you'd like

Cura already contains the ability to concisely display the differences between two profiles.  Currently, the only two profiles that can be compared and the differences displayed to the user are the version of the profile being edited in storage and the version of the same profile held in memory, just before committing the memory version to storage (the "Save Changes" button).

The solution I propose is to expand the use of Cura's ability to compare two profiles and display the differences to the user to any two profiles the user choses. 

Maybe:

File > Compare Profiles...

Which brings up a dialog box where the user selects two profiles (so maybe a file 1 field and a file 2 field) and presses a button "Compare."  Cura then compares *those* two profiles and displays the differences to the user, only now there is no "Save Changes" button.

### Describe alternatives you've considered

The only alternative is to export the two profiles of interest and compare them manually using another tool.

### Affected users and/or printers

I am guilty of this:  I tweak a profile trying to get a good print.  Before I know it, I don't know what I changed compared to the profile I started with.  This change will benefit anyone who finds themselves in the position I described.

GregValiant, 2025-03-19:
You can compare the changes to a profile to it's own "default" settings by reselecting it.  That brings up the "keep or discard changes" dialog which has a side-by-side view.

[image]

But I can see that the same sort of thing with two different profiles side-by-side would be handy.  Maybe that same dialog could be duplicated and a "Pick two" option added.

I keep my profiles to an absolute minimum.  Fast PETG, Slow PETG, Fast PLA, Slow PLA, and really slow TPU.  Every model gets it's own love so I see no need to have a "Fast PETG with Normal supports" and "Fast PETG with Tree supports", etc.  I just start out with a generic profile as a starting point and everything is custom for that model.  If I want to save the custom stuff I save it as a project file so I can pull it back up.  That's what works for me.

onolox, 2025-03-19:
There's a plugin outisde marketplace that do what you want.