Ultimaker/Cura #20412: Multi-Nozzle database like

EmEiBee, 2025-03-19:
### Is your feature request related to a problem?

Problem:
If someone constantly changes nozzles, and not just nozzles from the same manufacturer, meaning they have a significant Z-offset difference, CURA isn't exactly helpful.

### Describe the solution you'd like

Dirty solution:
Currently, I enter the nozzles used in the CURA printer settings with a 1/1000th decimal place, for example, "0.401" for a 0.4 mm nozzle, and set the Z-offset I determined for this exact nozzle with...

```
{if machine_nozzle_size == 0.201} M851 Z-3.25 ; Steel
{elif machine_nozzle_size == 0.301} M851 Z-3.03 ; Steel
{elif machine_nozzle_size == 0.302} M851 Z-3.15 ; Brass HF
...
```

This thousandth makes no difference when calculating the dependent parameters. At least I haven't noticed any impact so far.
However, it would be better if you could enter the 1/1000 place and evaluate it via IF/THEN, but the 1/1000 place would be ignored when calculating all dependent parameters

### Describe alternatives you've considered

Desired solution:
I would much prefer to have a corresponding option here to create a kind of database (profiles) with the corresponding nozzles. 
In my opinion, it would be even better if the nozzle size were included in the profile per printer anyway, i.e., transferred from the printer settings to the profiles, maybe as a variable, so that in the printer settings, instead of the actual nozzle size, I enter something like "$nozzlesize" into the field, whose value is populated with the actual nozzle size from the profile field from an option field under "Material" or "Quality." 
If there were also a "Nozzle Size" parent group parallel to "Material" (as was probably the case in the past), then in my opinion, that would be a huge step forward for those like me who constantly work with different nozzle sizes and different profiles.

### Affected users and/or printers

all user and all Printer with (quick) changeable nozzle

GregValiant, 2025-03-19:
I change nozzles constantly.  I see this as a niche request and I'm thinking it won't get a high priority.  There are other things you can do to get the effect without having to build a database.
{if machine_nozzle_size == 0.201}
M851 Z-3.25 ; Steel
{elif machine_nozzle_size == 0.301}
M851 Z-3.03 ; Steel
{elif machine_nozzle_size == 0.302}
M851 Z-3.15 ; Brass HF
{endif}
Will work in your startup as of 5.9.0.

That's what I do with steps/mm for various materials.
There is also variables "nozzle_id" which seems to just be used for UM nozzles, and there is also "machine_nozzle_tip_outer_diameter" that you could use as an alias.  I think either one  would need to go into the actual nozzle variant file.

EmEiBee, 2025-03-19:
You're probably right about the low priority. But I just wanted to mention that it's something that annoys me almost daily, especially when I've forgotten to change something somewhere.

> There is also variables "nozzle_id" which seems to

You've already helped me with the IF/THEN statement. But where can I find out which variables I can use where, and what variables even exist?
Is there documentation that even non-programmers can understand?

EmEiBee, 2025-03-19:
Thank you very much. That helps a lot.
I will try and see, what catastrophes I can create this way  :)

GregValiant, 2025-03-31:
I did get the "machine_nozzle_id" to work.
My installation uses the creality nozzles that are listed under the main "variants" folder (rather than the ones in the actual "variants\creality" folder).
I just had to add a line for the 'machine_nozzle_id' to each nozzle file.  The number is the Z-offset for the nozzle.
[values]
machine_nozzle_size = 0.4
machine_nozzle_id = -3.23

And in the Startup Gcode:
M851 Z{machine_nozzle_id}

That results in a line in the Gcode:
M851 Z-3.23

The model placement is a different issue and should be in a separate bug report.

Regarding this request, it appears that what you want to do is currently available within Cura.  Can I close this?

EmEiBee, 2025-03-31:
Well... great result. So if that works, you can close this..
Many thx. Things like that are the real frustration killer :)

closed 2025-03-31 by EmEiBee