Ultimaker/Cura #20413: Chinese coin_FU

My12123, 2025-03-19:
### Cura Version

5.10

### Operating System

Windows 10

### Printer

easythreed K9

### Describe model location

x-20 y20 z0

### Describe your model

-

### Add your .zip here ⬇️

[Chinese coin_FU.zip](https://github.com/user-attachments/files/19340918/Chinese.coin_FU.zip)

GregValiant, 2025-03-19:
Something is wrong with that upload and I can't extract the file.

GregValiant, 2025-03-19:
Alright, 7-zip was able to extract it.
The model is fine and sliced for me.  I can't duplicate the problem.  Use the "Help | Export File for Technical Support" command and zip that 3mf file and post the zip folder here.  It will have your cura.log file in it.

closed 2025-03-19 by My12123