Ultimaker/Cura #20415: Slicing failed after upgraded from Cura 5.6 to Cura 5.9.1

tbayart, 2025-03-19:
### Cura Version

5.9.1

### Operating System

Windows 10 x64

### Printer

Ender 3 v2

### Name abnormal settings

resetted all settings to default and restarted cura

### Describe model location

tried moving, rotating and scaling down the model

### Describe your model

successfully sliced under Cura 5.6 (before i update Cura)

### Add your .zip here ⬇️

[Kunai support.zip](https://github.com/user-attachments/files/19344235/Kunai.support.zip)
Initially i faced the issue with another new model i'm working on
I updated Cura to latest before slicing my new model and faced the error
So i tried slicing this model since i know it was previously working

HellAholic, 2025-03-19:
Based on the log you shared in the other ticket, you somehow managed to combine the two different versions.
The slice message is being sent back from the 5.6.0 version engine rather than the 5.9.1

[image]

as in comparison to:

[image]

My guess would be that you installed it on top of the previous version.

Regardless of how it happened, the generic fix would be:
1- Start with removing the previous installation (5.6 / 5.9.1)
2- Clear the configuration folders: (Warning - this will remove any printer instances or custom profiles so having back ups is recommended)
    A - %appData%/Roaming/cura
    B - %appData%/Local/cura
3- Install the 5.9.1 version. Make sure you don't install over any existing Cura versions.
4- Add your printer and import your custom profiles
5- If you still encounter the issue, there might be a problem in the custom profiles you can clear them using the step 2

tbayart, 2025-03-19:
> My guess would be that you installed it on top of the previous version.

Seems you guessed right, i thought msi installer removed the previous version but that is not true.

I uninstalled all installed versions and reinstalled Cura and it's working fine again.

Thanks @HellAholic for that quick resolution

closed 2025-03-19 by tbayart