Ultimaker/Cura #20420: Cura 5.10 BETA-1 - Overhanging walls problem with low-res geometry

sargonphin, 2025-03-20:
### Cura Version

5.10 BETA-1

### Operating System

Windows 11

### Printer

Custom Tronxy X5SA-330 Pro

### Reproduction steps

While testing the new BETA feature, I ran across some strange issue regarding overhang wall speeds. Overhangs seem to be broken down into small segments with large speed variations when geometry isn't high enough resolution

[image]
[image]

### Actual results

Causes the printer to jerk a lot (successive acceleration-deceleration) and of course results in bad print quality

Here are the settings I put for the overhangs. I more-or-less followed what is given in the example on the release page

[image]

### Expected results

Entire wall loop is considered overhang. I know this is actually a "fix" from previous version that would also print non-overhang walls in the same loop at low speeds, so this is an improvement over old version, but looks like there need to be some smoothing when the speed changes a lot over small segments

### Add your .zip and screenshots here ⬇️

Here is the project file to review the model and printer settings: [UCP_TubeDuctPP.zip](https://github.com/user-attachments/files/19374744/UCP_TubeDuctPP.zip)

GregValiant, 2025-03-20:
Thanks for the report.
I can't duplicate this using the UCP file.  If you would save a regular "Project File" and post it all your settings will be included..

[image]

sargonphin, 2025-03-21:
Okay let's try this: [UCP_TubeDuctPP.zip](https://github.com/user-attachments/files/19389192/UCP_TubeDuctPP.zip)

Let me know if this is better!

GregValiant, 2025-03-22:
Better.  This is the sliced project in 5.10beta

[image]

When I change to any other setting profile, and then make changes to it so it matches the "CoreXY-3-300 Normal" profile the speed oddity goes away.

[image]

When I switch back to the "CoreXY..." profile, it shows up again.

In this case the bug might be in how Cura put the settings into the "CoreXY..." profile rather than a particular setting that you are using.  Even when I disable the "Overhanging Wall" settings in the "CoreXY..." profile the problem persists.

The Cura team will take a look.  There may be a corruption in that profile.  I would suggest deleting it and putting a new one together and see if the problem goes away.  I've seen what appears to be "profile corruption" in other bug reports.

sargonphin, 2025-03-22:
Noooooo that was the last thing I wanted to hear!... 😆 Okay I will have a look and maybe try with some other profiles. This profile is really really old, dates back from early Cura 4.x versions. Could be that it needs to be refreshed into a new one indeed. I will let you know what happens. I will also attach the exported profile file that I use, maybe this could help debugging. It will have to wait since I'm not on my PC for the moment-- maybe later tonight!

sargonphin, 2025-04-05:
Fixed in Cura 5.10.0 stable (ignore the... really strange supports down there 🤔). I also disabled the "skin as bridge threshold" settings and it appears to produce better results with slicing quality

[image]

closed 2025-04-05 by sargonphin

GregValiant, 2025-04-05:
_"This profile is really really old, dates back from early Cura 4.x versions."_
It would not suprise me if that was the problem.  I have a script that I can run on different Cura versions to check for changes in the settings (how many, what's new, what's obsolete, etc.).

This is the result when comparing Cura 4.13.1 to 5.10.0:
ADDED to v5.10.0 (163)  for a total of 693 settings
OBSOLETE from v4.13.1 (41) and the total was 558 settings

So there is a big difference there and you may have gotten bitten by that.

sargonphin, 2025-04-05:
In fact I am using the same profile as before. Probably adjusted some setting, or maybe updating to latest version did the trick. I have a strong feeling that the supported skin threshold for bridge might have something to do with this, although previous contributor didn't have the same issue so... I guess this is a mystery!